NVIDIA-AI-IOT/jetracer #4: i18n contributions

tomo-makes, 2019-07-01:
Thanks for the amazing project and resources! I translated this repository resources (notebooks, markdown files) into Japanese. 
https://github.com/tomo-makes/jetracer/tree/20190630-ja

Could you guide me if there is a proper / better way to contribute (pull request under specific directories etc)?

Tomo

jaybdub, 2019-07-02:
Hi @tomo-makes,

Thanks for sharing this!  

Currently, we're not decided which contributions would be best merged upstream, or referenced in a fork.  We may not be able to maintain the language variants, so I think for now the best method will be for others to reference this issue and your fork.

This is similar to how we treated community contributed BOM's for JetBot, (see [this issue](https://github.com/NVIDIA-AI-IOT/jetbot/issues/12)).

Thank you!
John

xxzvxx, 2019-07-04:
on the wiki page none of the link are working... :(

> Follow these steps and you should be up and running with JetRacer in no time!

Get parts from the bill of materials
Follow the hardware setup
Follow the software setup
Run through the examples

>

tomo-makes, 2019-07-06:
Hi John - 

Thanks for your quick and polite response! Yes I understood the current maintenance policy.
Looking forward to seeing more exciting projects and enhancements coming up!

Tomo

closed 2022-07-18 by jaybdub