NVIDIA-AI-IOT/jetracer #6: Criteria for choosing the RC car

abhinav-upadhyay, 2019-07-07:
Hi,

This is not really an issue but a query. Did you choose the Traxxas LaTrax Rally car specifically for a particular reason, such as the 3d printed base board would fit well with it, or you knew that it was hackable? I am asking because the Traxxas RC cars are not available in India and importing it costs 3x more. There are other RC cars with similar specs (1:18 scale, 2.4 GHz radio, 4WD), such as [this](https://www.amazon.in/Sky-Zone-Waterproof-Controlled-Crawler/dp/B07JK3J437/). 

The Donkey car is also a relatively affordable option. I would appreciate if you could give any opinion on this, I have never played with RC cars and haven't done any hardware hacking before.

tokk-nv, 2019-07-08:
Hi Abhinav,
Thank you for your question.

We chose Latrax Rally car mainly because it was hackable and small (being small is nice when it comes to demonstrating in a small space).
But you should be able to use other cars, as long as **they expose Servo and ESC controls through the standard 3-pin "Servo" connectors**.
So anything used in Donkey Car should work.

Our observation is that, once the RC car gets smaller than 1/10 scale, it uses what people call 3-in-1 controller, which combines the radio receiver (RX), the servo controller and the ESC in one unit.
This makes it very difficult to hack, as you cannot simply plug the cables from Servo and ESC to the PCA9685 board.
You can tell the car is using 3-in-1 design by finding the 5-wire cable coming out from the servo, like the one on Amazon you suggested.
Availability of the upgrade/replacement parts is also a good indication that the car is hackable.

I think you will have a better chance if you go for 1/10 scale and look in the hobby section rather than the toy section. (Perhaps you can find some relevant info on a forum like [this](http://www.rcindia.org/index.php)?)
I hope you will find a reasonably priced car for your project, and please share with us if you find a good one.

jasonlililililili, 2020-10-25:
May I ask which component has signal input and power output? Because I have no idea which component (RC reveiver? PCA9685? Servo Motor Driver? RC Servo? or the RC ESC?) should be connected to the battery.
Thank You!

krissdap, 2020-11-17:
Battery sends power to ESC.

In normal RC car, ESC connects to RC receiver, sends the power, and the receiver shares the power the steering servo connected to it.

In JetRacer, you have the servo multiplexer in the middle. And you power the Jetson Nano with USB power bank.

Of course you can also power each component in other ways if you are adventurous. :)