NVIDIA-AI-IOT/jetracer #12: VCC or V+?

zlite, 2019-07-21:
In this step of the instructions, you say to connect to VCC but the picture shows the connection to V+. Which is right? 
https://user-images.githubusercontent.com/25759564/60315002-a56b1000-9919-11e9-8a53-2d941cc1b139.JPG

zlite, 2019-07-21:
If you plug it into V+ bad things happen! Nukes carrier board, wifi, etc. Yikes

jaybdub, 2019-07-21:
Hi Chris,

Thanks for pointing this out!  You're right, it's supposed to be VCC as stated.

We had this in right in an earlier shoot, I've updated to use the correct image.  You can find the change here

https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/hardware_setup.md#step-1---connect-jetson-to-servo-driver

Sorry for the confusion, I hope this didn't cause too much trouble.   Please reach out if you find anything else!

Thanks,
John

closed 2022-07-18 by jaybdub