NVIDIA-AI-IOT/jetracer #14: Can you use C++ and ROS to control the car

deeprun, 2019-07-24:
Seems you have to run Python via WIFI to control the car. What if you do not have WIFI? Can you run C++ and ROS to control the car?

jaybdub, 2019-07-27:
Hi Deeprun,

Thanks for reaching out!

>  What if you do not have WiFi?

You actually don't need WiFi.  This is a convenience so you can easily access the programming environment for rapid prototyping.  If you don't have WiFi, you can connect directly over micro USB, and access the web programming environment at ``192.168.55.1:8888``.  Or you could always open the a web browser on the Jetson Nano itself.  This would consume memory on the Nano though.

The examples are Jupyter notebooks (which require a web browser), but you can still operate the robot in plain Python (ie ``racecar.throttle = 0.3`` still works.  

> Can you run in C++ and ROS to control the car?

We don't have a C++ library or ROS interface for JetRacer.  To create one, the primary effort would be in interfacing with the ``PCA9865`` servo driver chip to control the steering / throttle.  Perhaps there are C++ libraries (or maybe even ROS) for this purpose.

Please let me know if you have any other questions.

Best,
John

UberWagen, 2020-08-14:
> We don't have a C++ library or ROS interface for JetRacer. To create one, the primary effort would be in interfacing with the `PCA9865` servo driver chip to control the steering / throttle. Perhaps there are C++ libraries (or maybe even ROS) for this purpose.

Old thread, but I'm in the process of trying it. [This guy](https://github.com/dheera/ros-pwm-pca9685) did write a ROS driver for the 9865 if anyone finds this in the future.

closed 2022-07-18 by jaybdub