NVIDIA-AI-IOT/jetracer #20: RC transmitter modification for tamiya version

hygradme, 2019-08-24:
Hi, I made JetRacer of tamiya version. 

Then, I followed the instruction in [docs/examples.md](https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/examples.md) Example 1 - Basic Motion.
When I run the line of car.steering = 0.3 in [notebooks/basic_motion.ipynb](https://github.com/NVIDIA-AI-IOT/jetracer/blob/dev-stl/notebooks/basic_motion.ipynb), my car doesn't response.
The comment in the notebook said:
> If the car does not respond, it may still be in ``manual`` mode.  Flip the manual override switch on the RC transmitter.

However, my RC transmitter ( which is TTU-08 also in the photo of [docs/tamiya/bill_of_materials.md](https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/tamiya/bill_of_materials.md)) doesn't have override switch.

It seems like the hardware setup for latrax (at the Phase 4 of [docs/latrax/hardware_setup.md](https://github.com/NVIDIA-AI-IOT/jetracer/blob/dev-stl/docs/latrax/hardware_setup.md)) has RC transmitter modification.
Does tamiya version also have to do same modification like latrax version to make it ``manual`` mode?

romybompart, 2019-08-26:
Hi. I was reading, it uses channel 3 switch to change from manual to "Nvidia" mode. 

Channel 3 is connected to SEL which is the input of the PIC10 at the PWM multiplexer. So that pin detects a pwn threshold to let channel M1 to M4 or S1 to S4 pass to OUT1 to OUT4 respectively. 

In this manual you will see what is channel 3 switch:

manuals.hobbico.com › futPDF
Resultados de la Web
1M23N31504 - You have reached ftp.hobbico.com

If this does not work for you you can by pass the multiplexer and connect the PCA PWM driver ch0 to the steering wheel servo and ch1 to the ESC, but you won't have command on the car by your transmitter.

Let me know whatever you got.

hygradme, 2019-08-26:
Thank you for your advice.

I think I didn't have enough knowledge about RC transmitter. 
My  [Tamiya RC Transmitter](https://www.amazon.com/Fine-Electric-Drive-Model-tamiya/dp/B0077K0WM2) has only 2ch which cannot send SEL signal to the receiver.

I bought combo version of NSX RC car with Tamiya TT02 and RC transmitter which is different from the one in [bill of materials](https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/tamiya/bill_of_materials.md).
I should have bought Futaba 3PV 3ch 2.4GHz transmitter.

As you told me, I bypass the multiplexer and switch "Nvidia mode" and "Manual mode" with just replacing the cable from RC transmitter with the one from motor driver by hand.
It is not convenient to replug cable every time after collecting data to switch to "Nvidia mode" but it works at least.

hygradme, 2019-09-01:
Sorry for late reply and thank you for your kind advice.
I followed your advice but I couldn't switch "Nvidia mode" and "manual" mode from ch2 of the driver.
As you mentioned, SEL may not receive valid signal from pca9685.
I cannot use oscilloscope right now so not sure the exact reason of this  but I will try the 2nd option which directly connects GPIO pins to SEL pin.

itsklee, 2020-06-20:
@romybompart @hygradme 

I find the  way 
pca96855 can  make signal  below

>   The multiplexer considers the RC signal on the SEL channel to be valid if it has a 10–330 Hz pulse 
>   rate and 0.5–2.5 ms pulse width. The other inputs (S1–S4 and M1–M4) are unrestricted.
> 

first follow @romybompart  's way
and  add some  code in 
   class NvidiaRacecar(Racecar):

    def __init__(self, *args, **kwargs):
        super(NvidiaRacecar, self).__init__(*args, **kwargs)
        self.kit = ServoKit(channels=16, address=self.i2c_address)
        self.steering_motor = self.kit.continuous_servo[self.steering_channel]
        self.throttle_motor = self.kit.continuous_servo[self.throttle_channel]
        self.manual = self.kit.continuous_servo[self.manual_channel]
        
        **self.kit._pca.frequency= 100
        self.kit._pca.duty_cycle = 0x7fff**

and  then set  the  value

   car.manual = 0  #  will  control by  your trasmitter
 
or 
   car.mamual = 1 #  will control  by your pca96855 
  
thank you

abdul-mannan-khan, 2021-05-03:
I am having the same issue. I followed this method and my race car is still not moving.

closed 2022-07-18 by jaybdub