NVIDIA-AI-IOT/jetracer #21: No module named 'jupyter_clickable_image_widget'

aq44634, 2019-09-06:
notebooks/interactive_regression.ipynb
### Data Collection
```py
import cv2
import ipywidgets
import traitlets
from IPython.display import display
from jetcam.utils import bgr8_to_jpeg
from jupyter_clickable_image_widget import ClickableImageWidget
```
```
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-4-ffabc01d44a0> in <module>
      4 from IPython.display import display
      5 from jetcam.utils import bgr8_to_jpeg
----> 6 from jupyter_clickable_image_widget import ClickableImageWidget
      7 
      8 

ModuleNotFoundError: No module named 'jupyter_clickable_image_widget'
```

I can't find a package or module named "jupyter_clickable_image_widget"

jaybdub, 2019-09-25:
Hi aq44634,

Thanks for reaching out!

Are you using the [JetCard](https://github.com/NVIDIA-AI-IOT/jetcard) image listed in the software setup instructions?  This package is pre-installed on that SD card.

Best,
John

hezhenjiang, 2020-06-28:
have the same problem .
how can we intsall "jupyter_clickable_image_widget" on xavia nx?

abritten, 2021-05-26:
Hi @jaybdub 

I got the error on my Jupyter Labs on VScode. I followed your setup and the only one that didnt work was:
    jupyter labextension install js

I restarted my VScode and now I got a registration problem with the widget. 
    Error: Module jupyter_clickable_image_widget, semver range ^0.1.0 is not registered as a widget module

How do I register it?

closed 2022-07-18 by jaybdub