NVIDIA-AI-IOT/jetracer #22: unable to train and run

gbr1, 2019-09-20:
Hi!
Just builded my Jetracer (latrax 1:18 -> [youtube video](https://www.youtube.com/watch?v=5ZBqgufzdm0) )
I created a setup track:
[image]
<br>
After following instructions:
1) _basic motion_ works correctly;
2) _interactive regression_ gives me always right-bottom blue point during "evaluation";
3) _road following_: car always go right.

I always tried with a simple path:
[image]
And car goes right.

Installed camera is: IMX219 with wide angle.

How do I choice correctly points?  Which is correct sequence (create images -> train button -> evaluate -> save)?
Is my track correct or are necessary other stuffs such as central line and uniform colours?

Thank you in advance!

jaybdub, 2019-09-25:
Hi gbr1,

Thanks for reaching out and sharing your video!  Very well done!

The track you created looks good to me, should certainly be able to work on that.  Hopefully we can get you up and running soon.

Could you provide the following information

1.  How many data samples did you collect / how was car placed?  (data variance is very important)
2.  How many epochs did you train for?  (it usually takes many, I'd guess at least 15+ to converge)

Best,
John

gbr1, 2019-09-28:
Hi @jaybdub!
I took about 200 photos and 3 epochs maybe that was the problem.
Then I moved on donkeycar software so I didn't try more.

Thank you.

RyanC1681, 2019-11-12:
Hi, John:
I took >200 photos but epoch always = 0 and car always do circle, not steering straight.
Could you give me a correct training sequence (live -> train button -> click left image -> count# inceasing -> evaluate -> save model)?  
How to set another epoch or find out which point need more training/images?

Thanks

jaybdub, 2019-11-13:
Hi RyanC1681,

The work flow is generally to 

1.  Collect images
2.  Train for several epochs until loss converges
3.  Run live to visualization how well the network is performing
5.  Repeat 1->3 until you are happy with the performance, at this point save the model

Please let me know if this helps.

Best,
John

ctye85, 2020-01-10:
Hi jaybdub,

I know this is a bit old, but perhaps you could help me out with my current situation if you have time.

I can get to the point where I'm consistently training a model that performs well after 100+ images and 15 or so epochs. It tracks movements around the track with a high degree of accuracy.

I essentially do exactly the work flow you described above: collect images, train for several epochs, and run it live to make sure the model is performing well. Rinse and repeat, then save the model when I'm satisfied.

However, when I attempt to run it with the Road Following notebook, it just turns in a continuous circle to the right. It does seem to adjust the degree of turning at times, which indicates that some commands seem to be going through.

Do you have any idea why this may be happening, considering the perceived accurate performance running the Interactive Regression notebook?

Thanks!

closed 2022-07-18 by jaybdub