NVIDIA-AI-IOT/jetracer #24: any replacement for servo multiplexer?

nelsonjin, 2019-09-27:
Is there any replacement for servo multiplexer? 
I can't buy it in my country...

tokk-nv, 2019-09-27:
Hi nelsonjin,
Unfortunately, as far as I'm aware right now, we don't find any good alternative to the Pololu servo multiplexer.
The servo multiplexer is just to add a safety feature, which allows you to stop the car when the computer or your code goes into an uncontrollable state.
So you can choose to just connect the PWM outputs from the PCA9685 breakout board directly to your servo and the ESC on your car, if you don't need that feature.

nelsonjin, 2019-09-29:
thanks, I'll try this with cautious

cygnusyang, 2019-11-08:
> Hi nelsonjin,
> Unfortunately, as far as I'm aware right now, we don't find any good alternative to the Pololu servo multiplexer.
> The servo multiplexer is just to add a safety feature, which allows [...]

Hi tokk-nv,

I am trying to use nano to control the servos without the bluetooth receiver board. I found that the servo only can be controlled if I connect the bluetooth receiver board first. It seems that the bluetooth receiver is designed to negotiate with the servo control board.  I have tried to use oscilloscope to get the PWM output, the PWM output from PCA9685PW chipset is the same as CC26XX from the bluetooth receiver (about 100HZ). The GND and VOL(3.4v) also are the same. Do I need to send some signals to the servo control board before the PWM? Thanks.

larryw-jimi, 2022-02-18:
do we have any replacement now