NVIDIA-AI-IOT/jetracer #27: torch2trt  no attribute for reciprocal_

nelsonjin, 2019-10-11:
ERROR : AttributeError: module 'torch' has no attribute 'reciprocal_'

while running 
'''
from torch2trt import torch2trt
data = torch.zeros((1, 3, 224, 224)).cuda().half()
model_trt = torch2trt(model, [data], fp16_mode=True)
'''

is there any version problem for torch2trt? 

[image]

jaybdub, 2019-10-11:
Hi nelsonjin,

Which version of PyTorch is this using?  I believe using version 1.2 would resolve this issue.

That said, this error seems like an issue in torch2trt that should be addressed.  I will look into this.

Please let me know if you have any additional questions.

Best,
John

nelsonjin, 2019-10-12:
torch is preinstalled by "jetcard_v0p0p0.zip", which is installed by jetcard.  version : 1.1.0a0+b457266

`wget https://nvidia.box.com/shared/static/veo87trfaawj5pfwuqvhl6mzc5b55fbj.whl -O torch-1.1.0a0+b457266-cp36-cp36m-linux_aarch64.whl
echo $password | sudo -S pip3 install -U numpy torch-1.1.0a0+b457266-cp36-cp36m-linux_aarch64.whl`

nelsonjin, 2019-10-12:
Solved by installing torch-1.2.0

closed 2019-10-12 by nelsonjin

ilhmfahmi, 2021-01-16:
how to installing torch 1.2.0?

ilhmfahmi, 2021-01-26:
> Solved by installing torch-1.2.0

How to install torch-1.2.0?