NVIDIA-AI-IOT/jetracer #34: error message after running Jetracer code

goodeyedeer, 2019-11-01:
show error message as below, please kindly help

[image]

jaybdub, 2019-11-13:
Hi goodeyedeer,

Thanks for reaching out, and sorry for the delayed response.  This should be resolved by updating the version of PyTorch to use version 1.2.

You can find it here

https://devtalk.nvidia.com/default/topic/1049071/jetson-nano/pytorch-for-jetson-nano-version-1-3-0-now-available/

Best,
John

TimIWT, 2020-09-09:
> Hi goodeyedeer,
>
> Thanks for reaching out, and sorry for the delayed response. This should be resolved by updating the version of PyTorch to use version 1.2.
>
> You can find it here [...]

Hi John.

I installed the new PyTorch version, but afterwards i cant install ResNet, because the Download stops at around 80% for some reason. Is this issue known?

Best regards,
Tim

closed 2022-07-18 by jaybdub