NVIDIA-AI-IOT/jetracer #36: Use of Futaba S3004 Servo in  Tamiya Setup

baburaj2019, 2019-11-03:
Futaba S3004 Standard Ball Bearing Servo is listed in Tamiya Bill-Of-Materials . But its not used in the Hardware assembly . 

So is it  Futaba S3004 Standard Ball Bearing Servo  really required for Tamiya setup ?

chitoku, 2019-11-06:
Hi baburaj2019,

You can use other servos, as long as it works on the Tamiya platform.
The Futaba servo is not shown in our Hardware Assembly, because it is part of Tamiya TT-02 assembly process.

jasonlililililili, 2020-10-10:
Follow up, does it mean that if the RC car that I buy has servo for front wheel turning originally, I do not need to buy that S3004 servo?
Or in other words, is the S3004 servo for turning direction of the front wheels of the RC car? Thanks!

krissdap, 2020-10-18:
@jasonlililililili you are correct.
Any hobby servo with standard connector, and fit the RC car you are going to use, should work.

closed 2022-07-18 by jaybdub