NVIDIA-AI-IOT/jetracer #39: Tamiya version problem

xingtongli, 2019-11-23:
Hi, 
I made a tamiya version. When I tried the basic motion part, the throttle command did not have any response from the car. I viewed an issue about the transmitter do not have manual override switch. So I bypass the multiplexer and connect the cable of servo motor and ESC to the PCA PWM driver directly. The steering command had response but the throttle did not have. My ESC's model number is TAMIYA TBLE-02S. What problems it might have? Thank you.

Ziegan, 2019-11-25:
I am also facing the same problem, while using the TAMIYA TBLE-02S ESC.
When bypassed to PCA PWM directly the steering command works....and the throttle dose'nt respond.
Is there any way to overcome this issue ?

Ziegan, 2019-11-25:
I have a TAMIYA build and all its necessary components that are stated in the billing informations for TAMIYA build.
My TAMIYA includes TBLE-02S ESC to control the motor.
The car doesn't respond while trying out the example basic motion,
So i tried out it with a different ESC - Jreleas F540 sensor-less motor ..., which also didn't work, 
but when bypassed to the PCA PWM directly, the steering command works well with both the ESC's but the motor doesn't work.
Is there anyway to make this work ? 
thanks.

Ziegan, 2019-11-25:
I have a TAMIYA build and all its necessary components that are stated in the billing information for JetRacer (TAMIYA build).
My TAMIYA includes TBLE-02S ESC to control the motor.
The car doesn't respond while trying out the example basic motion,
So i tried out it with a different ESC - Jreleas F540 sensor-less motor ..., which also didn't work,
but when bypassed to the PCA PWM directly, the steering command works well with both the ESC's but the motor doesn't work.
Is there anyway to make this work ?
thanks.

Ziegan, 2019-11-25:
Hello 

**Goal** : I am Trying to Build JetRacer based on Tamiya Version . 
I have bought all the Materials Mentioned in the Bill of Material of Tamiya based JetRacer 

**Problem**: When Tried the Basic Motion, After making all the connections as per the  Build Instructions, The Car didn't Respond 

**Options I have Tried**:
My TAMIYA includes TBLE-02S ESC( Speed controller)  to control the motor.The car doesn't respond while trying out the example basic motion,
So i tried out it with a different ESC - Jreleas F540 Brushless motor ..., which also didn't work,  

When the  Polulu Multiplexer is not used when bypassed to the PCA PWM directly, the steering command works , But the Motor doesn't respond 
Is there anyway to make this work ?

Ziegan, 2019-12-10:
Still waiting for any updates to overcome this problem !

jaybdub, 2019-12-11:
Hi All,

Thanks for reaching out, and terribly sorry for the delayed response!

Could you try inverting the throttle gain and then setting the throttle?

Perhaps,

```python
car.throttle_gain = -0.5
```

and then

```python
car.throttle = 0.5
```

As a safety precaution, make sure the car is suspended in the air.  It can go fast.

Please let me know if this works or not!  If it resolves the issue I will consider how to make this point clear.

Best,
John

tokk-nv, 2019-12-11:
Hi all,

In addition to what John suggested, it might help to put the motor ESC into Reverse Enabled mode.
It is possible that you are outputting PWM pulse to just apply "break" if the ESC is set to Reverse Disabled mode, so you want to flip the throttle_gain positive/negative in that case.

You can refer to TBLE-02S ESC manual for enabling reverse.
https://www.tamiyausa.com/media/files/45057ml-801-efb9.pdf

Ziegan, 2019-12-18:
Thanks for Your Help Guys..!
Finally my Tamiya Setup is working, it needs the throttle set up in negative for the car to move forward and positive to stop .., but my car can't get reverse !

Ziegan, 2019-12-28:
Recently,

I made another tamiya build using the list of materials shared in this page..,

While executing the program (basic_motion),
 i got this error...

Any suggestions to overcome this error

[image]

jaybdub, 2020-01-10:
@Ziegan 

Glad to hear!  Did you set the ``throttle_gain`` to a negative value?

Did you follow the TBLE-02S ESC manual that @tokk-nv mentioned in previous comment.

Best,
John

itsklee, 2020-01-16:
Hello.
I'm making Tamiya version
The same thing is happening with Ziegan's problem.
Of course, jaybdub's comments were handled.
Negative processing of throttle_gain was applied, as well as the setting of Tamiya receiver.

By the way, when throttel is negative, moving forward,
If it is positive, it is operating as stop.
It does not work backwards.

Is there another way?

itsklee, 2020-01-19:
> 여보세요. 
> 나는 타미야 버전을 만들고 
> 있습니다. Ziegan의 문제와 같은 일이 일어나고 있습니다. 
> 물론 jaybdub의 의견이 처리되었습니다. 
> 타미야 수신기 설정뿐만 아니라 throttle_gain의 부정적인 처리가 적용되었습니다.
> 
> 그건 그렇고, throttel이 음수 일 때 앞으로 나아가고, 
> 양수이면 정지로 작동합니다. 
> 거꾸로 작동하지 않습니다.
> 
> 다른 방법이 있습니까?

i’m found the solution
reverse move need more throttle gain
because receiver setting is forward full but reverse half or less

thank you

joessu, 2020-06-20:
I set my throttle gain to -1.0 to see what raw values would get me. 0.2 which is -0.2 was dead stop, no matter what the car was doing. Anything higher was move forward, anything less was move backward. Reverse is enabled on the speed controller.

closed 2022-07-18 by jaybdub