NVIDIA-AI-IOT/jetracer #40: Receiver version

mysliwiec-tech, 2019-11-29:
Hello,
Is there any reason of buying RX TX Combo (Futaba 3PV) with R304SB as we are being pointed by moving to Amazon listing? As far as I know R203GF just don't have the voltage telemetry, which I don't think we use anyway, but it can save some extra bucks.

jaybdub, 2019-12-11:
@tokk-nv

tokk-nv, 2019-12-11:
Yes, you can use any receiver/transmitter with 3CH or more.

We have not fully tested everything, but this worked when I briefly tested it.
https://www.amazon.com/gp/product/B07PKNVZZ9/

mysliwiec-tech, 2019-12-19:
@tokk-nv  Thank you for your previous answer. It clarified  things a bit. I understand that one channel is for throttle, one for steering, but for what is the third one? And further on that does it require the controller to have anything else than just a throttle and steering capability (don't get for what is the 3CH)?

tokk-nv, 2019-12-20:
So we are demonstrating the use of the controller/transmitter's CH3 to control the Pololu MUX (multiplexer) for safety.

When CH3 is toggled to be OFF position (or ON, depends on how you interpret), the MUX is to take the steering (CH1) and throttle (CH2) signals from RX (radio receiver) so it is manual mode, but when CH3 is in ON position the MUX is to take the CH1 and CH2 signals from Jetson Nano (via PCA9685) so it is like Autopilot mode.

We find this to be very effective, as CH3 basically functions as a KILL switch when your "autonomous" car goes out of control.