NVIDIA-AI-IOT/jetracer #41: Receiver cable orientation

ctye85, 2019-12-19:
Hello, I have a question about the cable orientation in the final build.

The pictures in the build show the original 2 channel + B receiver, however it's unclear where the cables are supposed to be connected in the final build.

I can get the car running in Jupyter Lab, but when running the Basic Motion file, steering seems to control the car's throttle, and vice versa for throttle controlling the steering. In manual mode, the car controls as you would expect, this is only a problem for "Nvidia" mode.

Do you have any photos of the updated final build that would make the cable orientation more obvious? I feel like I must simply have wires in the wrong place or something.

Thanks a bunch!

tokk-nv, 2019-12-24:
Hello ctye85,

Given that you can correctly operate the vehicle in Manual mode, it sounds like you just need to flip the cables between your PCA9685 board and MUX board.

This shows the connection, and please note the channel 0 of the PCA9685 goes to S1 of MUX board, and channel1 of the PCA9685 goes to S2 of MUX board.
https://user-images.githubusercontent.com/25759564/62125310-a1684000-b281-11e9-99de-f458e6ca6b06.png

ctye85, 2019-12-24:
Thanks for clarifying. I actually tried this shortly after posting my original message and it worked like a charm. Cheers!

ctye85, 2019-12-25:
[image]

Thanks for your advice. Strangely enough, I had the cables in the orientation in your picture, but when I switched them to what's in my picture right now it actually works correctly! Even though it's reversed from what the instructions say.

Very curious...I wonder why that's the case. Anyway, your advice for flipping the cables still worked!

ctye85, 2020-01-03:
Hello again, I have a quick question about overall cable orientation.

It seems several things in my setup only work when they are done in reverse of what the GitHub page says.

For example, the above-mentioned channel 0 going to S2 and channel 1 going to S1 on the MUX. This is strange but it works, so I can't complain too much I guess.

Additionally though, Ch3 on the RC receiver goes to SEL on the MUX as usual. However, it works correctly if I connect CH2 on the RC receiver to M1 on the MUX, and CH1 to M2 on the MUX. Can you provide any insight as to why this may be the case? If it works then I'll keep it that way of course, I am just very confused about why that's the case.

Could it possibly be a hardware update issue, like since you made the build the MUX from the manufacturer has been slightly changed or something?

Thanks for any additional info you can provide!

closed 2022-07-18 by jaybdub