NVIDIA-AI-IOT/jetracer #42: Issue running basic_motion.ipynb directly after software setup

ctye85, 2019-12-27:
[image]
[image]

Forgive my ignorance, as I'm sure the answer to this issue is incredibly simple.

I have attempted to run basic_motion.ipynb after finishing the software setup. However, there are apparently some erroneous values in the code, and I am not experienced enough to figure out what to do to fix it.

I figured posting this as an issue would help others who get stuck right after the software setup as well.

Thanks for any advice you can give, and happy holidays!

Ziegan, 2019-12-28:
Recently,

I made another tamiya build using the list of materials shared in this page..,

While executing the program (basic_motion), right after the software setup
i got this error.

Any suggestions to overcome this error

[image]
[image]
[image]

Thanks

krissdap, 2019-12-29:
I had this problem.
Fixed it by running this command once on a terminal, then restart the notebook.

<code>
sudo pip3 install Adafruit-PureIO --upgrade
</code>
It'll upgrade Adafruit-PureIO package to 1.0.4 from 0.2.3 which resolve this problem.

credit to tsteinle at https://devtalk.nvidia.com/default/topic/1068733/jet-racer-basic-motion-error/?offset=2

ctye85, 2019-12-29:
Thanks for the suggestion krissdap, I'll give it a shot when I can and report back!

ctye85, 2019-12-31:
Coworker tried that out on our car at work and it fixed everything. Thanks again for the advice!

closed 2019-12-31 by ctye85

krissdap, 2020-01-01:
Glad it helps!
Thanks for reporting back.