NVIDIA-AI-IOT/jetracer #43: Throttle reverse issue in basic_motion.ipynb

ctye85, 2020-01-03:
Hello all, happy holidays!

I am having a strange issue when running the JetRacer basic motion notebook.

When I run basic_motion.ipynb and adjust car.throttle, positive values actually make the car go in reverse, and negative values move it forward.

If I adjust car.throttle_gain to a negative number, that allows positive values in car.throttle to move the car forward.

Basically, things seem to be working in reverse for some reason. Anyone have any clue what's going on here?

Let me know if you need additional info to give an answer and I'll provide what you need. Thanks!

jaybdub, 2020-01-10:
Hi cyte85,

Thanks for reaching out!

Do you mind sharing which car you're using (Latrax or Tamiya)?

I think one of those may require a reversed throttle gain.  Perhaps I should document this better if it is the case.

Best,
John

ctye85, 2020-01-11:
Hey John,

Thanks for the response!

I'm using the Tamiya TT02. As you mentioned, I have to use a reversed
throttle gain in the NvidiaRacecar class, which allows me to move forward
with positive values in the notebooks. And vice versa, a positive value in
the class means I have to use negative values for forward motion. Simple
enough, just confusing at first.

Since I'm fairly new to Python, I can't see what the potential problem
would be going from an apparently working model in Interactive Regression
to not working at all in Road Following. The OP who is using a Latrax I
believe said the blue circle in the live feed would always be in the
bottom-right corner for them, but I didn't have that problem. It tracked
along the path I clicked on rather well, actually. So my initial thoughts
are that it could be a problem when the file gets optimized through
torch2trt.

As I said, I'm very much a beginner, so please tell me if you think I'm off
base on that assumption.

Let me know if you need any additional info, I'll provide all I can. Thanks!

Best,
Chris

ctye85, 2020-01-11:
Hi John,

I'm using a Tamiya. I actually just set a reversed throttle gain and was
able to get it running smoothly with positive numbers for acceleration, so
problem pretty much solved. I'll close the issue, thanks!

closed 2020-01-11 by ctye85