NVIDIA-AI-IOT/jetracer #45: Problem with NvidiaRacecar class

NolanGC, 2020-01-11:
[image]

In both the basic_motion and road_following notebooks, the following line results in a timeout error:
car = NvidiaRacecar()

This same code ran fine a few weeks prior and we haven't changed anything since.

uod-jhardy, 2020-01-25:
Had the same problem, try:

sudo pip3 install Adafruit-PureIO --upgrade

upgrades from 0.2.3 to 1.0.4, the problem was solved for me

NolanGC, 2020-02-01:
Our Adafruit-PureIO version is also 1.0.4. Is it possible it is a hardware issue?

uod-jhardy, 2020-02-01:
Hi,
It could be but look at the end of your original problem posting. In the last error block it clearly shows "Adafruit_PureIO-0.2.3.py3.6.egg". I would still look there first.