NVIDIA-AI-IOT/jetracer #46: Issue compiling basic_motion.ipynb

risingwisdom, 2020-01-11:
After following all the software setup, when I try to run the first line in basic_motion.ipynb, I get this error:
[image]

[image]

saying an integer was required but got a type bytearray.

Not sure what to do to fix it.

ctye85, 2020-01-12:
https://github.com/NVIDIA-AI-IOT/jetracer/issues/42

I asked this a little while ago, and got some help. You just have to upgrade Adafruit-PureIO package from 1.0.4 to 0.2.3. This command in the terminal should do it:

sudo pip3 install Adafruit-PureIO --upgrade

risingwisdom, 2020-01-14:
That made it work! Thanks.

closed 2020-01-14 by risingwisdom