NVIDIA-AI-IOT/jetracer #48: trouble compiling in road_following.ipynb

risingwisdom, 2020-01-28:
After running through the code in interactive_regression.ipynb and training/evaluating a dataset, I am unable to get past creating the camera class in road_following.ipynb. Specifically, the block of code starts compiling but never finishes compiling.

Particularly I am talking about this block:
[image]
Everything before it compiles with no problem.

I also have no problem creating the camera class in interactive_regression.ipynb so I am unsure what would be causing it to hang in road_following.ipynb.

Any help would be appreciated. Thank you.

kyung-il-Jang, 2020-04-06:
I am suffering from the same phenomenon. Did you solve it?

risingwisdom, 2020-04-06:
I found that you can run that camera code first then everything else in order and it would work. As for why that matters (because it shouldn't) I still haven't found why. But you can still run the car that way.

kyung-il-Jang, 2020-04-06:
thank you for answer me, but i don't understand camera code first. 

you mean, run this code

from jetcam.csi_camera import CSICamera
camera = CSICamera(width=224, height=224, capture_fps=65) 

and then, the rest of the code in order?

kyung-il-Jang, 2020-04-07:
Thanks a lot, it worked. But, It cannot follow the line properly. ha, ha.
It's frustrating because it doesn't work well with 400 photos and 10 trains.

mladen-simplifa, 2020-06-07:
Hi.
Did anyone found the cause or a fix for this? I am facing the same issue.

MaxKru, 2022-03-09:
i have the same issue