NVIDIA-AI-IOT/jetracer #51: Replacing the Pololu 4-Channel RC Servo Multiplexer

zaher88abd, 2020-02-16:
Hello,

I am wondering if I can replace Pololu 4-Channel RC Servo Multiplexer with other chips. It is expansive to ship it to Canada. I don't have any experience with electronics, I appreciate any help.

krissdap, 2020-02-19:
Hi,

You can save cost by skipping the servo multiplexer but you will lose remote kill-switch as mentioned  here
[https://github.com/NVIDIA-AI-IOT/jetracer/issues/24#issuecomment-536114889](url)

zaher88abd, 2020-02-19:
Thank @krissdap for replay. Is that mean I can not connect TX RX Combo controller?

krissdap, 2020-02-20:
If you don't mind, you can manually connect-disconnect the servo and ESC wires, changing between your RX and the PCA9685.

That's the purpose of the servo multiplexer.

fariz, 2020-03-24:
yes i can t find it polulu multiplexer in my country, can you advice is there any replace component for polulu 4 channel

YaBr11, 2020-06-17:
Any Multiplexer with 2+ channels should work.

Keep in mind that the multiplexer also adds a safety level. If the ESC and the servo are "hard" wired to the PWM board without the Mulitplexer and the program crashes, the PWM will be continued with the last value, meaning the Jetracer driving uncontrollably with the last throttle and steering value.

That's why I would recommend using the multiplexer not only as a convenient feature, but also as a safety level in case of a software failure.

closed 2022-07-18 by jaybdub