NVIDIA-AI-IOT/jetracer #52: Car steering is reversed

TCIII, 2020-03-22:
I am using a Traxxas 4WD Slash RC Car in place of the Tamiya RC Car and have found during the basic motion tests that the Slash car steering direction is the opposite of the Tamiya car steering direction ie: the Slash steers left when steering is set to 1 instead of steering right.
Therefore I would appreciate it if someone can point me to the proper python code in jetracer where I can reverse the steering input so that my Traxxas car will steer right when steering is set to 1 instead of to the left.
TCIII

TCIII, 2020-03-23:
After reading a closed issue about how to correct the throttle setting as to forward/reverse, I noticed that the steering gain was set at -0.65 so I changed it to 0.65 and the steering now goes right with a "1" input and left with a "-1" input. It took a little bit of time to notice that I needed to have "code" set before saving the changes, but all is well now.

closed 2020-03-23 by TCIII

CamRoni1339, 2023-10-07:
@TCIII 

Hello, I am having the same problem where my RC car goes left on 1 input and right with -1 input. Could you elaborate and maybe be a little more specific on the process you went through to fix this problem? For example what do you mean by you needed to have "code" set before saving the changes? This would help me out so much.

Thank you

Cameron

TCIII, 2023-10-08:
@CamRoni1339,
It has been a long time since I worked with the Jetracer code.
When I said "you needed to have the 'code' set before saving the changes" I believe I had to make the negative value positive, save the change and then restart the docker program.

TCIII

CamRoni1339, 2023-10-08:
@TCIII I figured it out. Thanks for the help.