NVIDIA-AI-IOT/jetracer #53: BLE comm.  between JetRacer and Phone App

Sunggu-Kim, 2020-03-31:
Hi, This is SG Kim,

I hope to send some data from Phone App to JetRacer based on BLE comm.
Can you guide what to do to enable BLE comm between JetRacer and Phone App based on Python?
( JetRacer : device ( Peripheral ), Phone  : Central )

I tried to install pybluez, but failed it.  

Thanks in advance for your help.

kyung-il-Jang, 2020-04-13:
May I ask you a bit about jetracer road_following?

Sunggu-Kim, 2020-04-20:
> May I ask you a bit about jetracer road_following?
Hi, KI Jang, Sorry... I saw this today.
What's it if you have still it?

kyung-il-Jang, 2020-04-21:
Hi, thanks for answering.
Did the road following work well with the default code?
My car continues driving to the right.
So, I change the bias value, but it continues.
Are there any precautions?

Sunggu-Kim, 2020-04-23:
Ahh... I also saw the same thing.
But, I didn't try to do any tunning.
I am focusing on other my own function.

Sorry...

SG Kim

closed 2022-07-18 by jaybdub