NVIDIA-AI-IOT/jetracer #56: Jetracer AI KIT nano  -——road_following problem

joker-wutong, 2020-04-17:
When I run the road_following instruction, the trajectory of the car movement is very different from the trajectory I want. How can I optimize the model used by road_following?

[合并视频20200409115735.zip](https://github.com/NVIDIA-AI-IOT/jetracer/files/4490694/20200409115735.zip)

chealyte, 2020-07-01:
you got it to work ? i try mine it stay still in 1 spot. I can control with ps2 controller that only thing work for me.

joker-wutong, 2020-10-26:
> you got it to work ? i try mine it stay still in 1 spot. I can control with ps2 controller that only thing work for me.

Yes, I have realized the function of using the jetson-Nano smart car to complete the tracing function. For how to do it, you can refer to the official tutorial published by NVIDIA on GitHub.

closed 2022-07-18 by jaybdub