NVIDIA-AI-IOT/jetracer #57: Jetson Nano B01

zaher88abd, 2020-04-18:
Hello,

I got Jetson Nano B01, and I tried included images in this repository but it didn't work. The Nano didn't boot at all. 
This there may you mention what this image has over the original image. So, I can run the code over this Nano jetson.

red24dog, 2020-04-19:
I successfully booted the Dec 13/19  Jetpack 4.3 image (NOT a Jetracer) from [https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write-linux] on a B01.  See this forum dialog: [https://forums.developer.nvidia.com/t/jetson-nano-new-version-b01/111680]. I believe the jetracer image has not been released that is compatible with B01, but nvidia should confirm. You may need to manually build the jetracer on this booted image until then.

zaher88abd, 2020-04-20:
Thanks @red24dog, for replay. 
Do you know what library, should install besides jetracer.

red24dog, 2020-04-21:
There does not appear to be any “ Install Jetracer from Scratch” documentation, as there is for Jetbot. However the Jetracer software setup describes steps to install code that includes a required Adafruit Library, from various NVidia github and PyPI. I believe this procedure assumes that a Jupyter Notebooks installation is the boot image and I am unsure if Jupyter Noteboks is in the Jetpack 4.3 image. I’m away from my workshop right now, but could check my B01 /Jetpack 4.3 device installation in next couple days and advise. As I hope also the forum referring to B01 release, has several posts requesting NVidia to release a B01 Jetracer image.
.

zaher88abd, 2020-04-21:
Thank @red24dog.
I will try that. And I hope @jaybdub, or @tokk-nv  give us a hint about the process.

red24dog, 2020-04-21:
I have to give a shoutout to the NVidia folks, dusty-nv, jaybbub & tokk-NB and others who have done great work with the Nano and it’s  support documentation for Jetbot, Jetracer, and all the AI tutorials.They are dealing with our  current worldwide  health situation like the rest of us and understand being unable to be as responsive as I’m sure they would like to be.

I checked my Nano B01 install from the Kit Getting Started, and there does not appear to be a Jupyter Notebooks installed nor running, though perhaps I accidentally did that.

 https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write

 So I would suggest you look at the Jetbot Software Install page and do steps 2-12 (to install AI packages, Jupyter Notebooks, various  Services), except step 9 that installs the Jetbot dependencies which you do not need nor want. https://github.com/NVIDIA-AI-IOT/jetbot/wiki/Create-SD-Card-Image-From-Scratch

Install the Jetracer github an its dependencies from its github.com instead of Step 9. Then you should hopefully be able to execute the Jetracer Notebook scripts and have fun.

I have NOT tried this post procedure so I apologize  if it doesn't work exactly;(

Good luck!

Sent from Ross Lunan

tokk-nv, 2020-04-22:
Hi all,
I'm sorry for the delay in response.

We have not yet come out with the JetCard image based on JetPack 4.3, however, a community member has come up with the SD card setup procedure.
(Yello highlighted lines are something you need to modify)
https://faboplatform.github.io/JetracerDocs/05.%E3%82%BD%E3%83%95%E3%83%88%E3%82%A6%E3%82%A7%E3%82%A2%E3%81%AE%E8%A8%AD%E5%AE%9A/01.install/?fbclid=IwAR1sVlX1ALBRxV52RsKheT9m1zecb-2oV2fi5mlInInodLzZJNVWSc2zSxg

We have not yet verified this nor pulled this into our script, however I'm hopeful this would just work.
It would be great if you can check this modified script and test on your B01.

Thank you for your cooperation in advance,
Chitoku

ARLunan, 2020-04-22:
Chitoku, thank you for that reference. Looking at the Jetracer Software Install, https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md, I believe FaBo procedure completes Steps 1 - 4? Is it then necessary to follow Steps 5-8 to complete the Jetracer Software installation of the jetracer dependancies  (Jetcam, torchRT, Jetracer packages)?

bjkayani, 2020-04-23:
I recently installed jetracer manually on the B01. I used the branch 4.2.2 and installed it via the instal.sh. It had errors that i solved one after the other and got it to work. Although now that I try to run the examples, the camera wont load. So I might try rebuilding from scratch as mentioned above.

bjkayani, 2020-04-23:
> Chitoku, thank you for that reference. Looking at the Jetracer Software Install, https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md, I believe FaBo procedure completes [...]

Yes, you still have to install those.

tokk-nv, 2020-04-23:
Like bjkayani answered, yes.
FaBo page actually has following sections to cover from Step 5.
https://faboplatform.github.io/JetracerDocs/05.%E3%82%BD%E3%83%95%E3%83%88%E3%82%A6%E3%82%A7%E3%82%A2%E3%81%AE%E8%A8%AD%E5%AE%9A/02.package/

bjkayani, 2020-04-23:
> Chitoku, thank you for that reference. Looking at the Jetracer Software Install, https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md, I believe FaBo procedure completes [...]

I can confirm that the changes from Fabo do indeed work on B01 without any glitched. Although step 8 for Fabo is different since their hardware is different so keep that in mind. Hope it works for you. I cant wait to train some models.

joessu, 2020-06-16:
I followed the instructions in those links and it seems the cuda version is newer than what install.sh pytorch install instructions can work with. the pytorch website has a little configuration form that gives you the command to use based on cuda version. https://pytorch.org I used sudo pip3 install torch torchvision. will hopefully verify it this weekend

pdemarco925, 2020-07-02:
Since some of you appear to have successfully created an image, is it possible to snapshot it and post it for others to use?

closed 2022-07-18 by jaybdub