NVIDIA-AI-IOT/jetracer #59: Wide angle Leopard camera replacement

jetrace, 2020-05-02:
Hello, I have a question about the setup components for Latrax-version JetRacer. Is it possible to replace the wide angle Leopard camera with a jetson nano compatible wide angle camera, e.g., 
https://www.amazon.de/IMX219-160IR-Camera-Developer-8-megapixel-Infrared/dp/B07TB8STRK/ref=sr_1_5?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=Wide+Angle+Jetson+Nano&qid=1588452355&s=computers&sr=1-5. Thank you very much.

mattkanwisher, 2020-05-16:
Whats annoying the project recommends 145 degree focal length which is impossible to find. 

It also seems like the h145 model is discontinued https://leopardimaging.com/product/nvidia-jetson-cameras/nvidia_nano_mipi_camera_kits/li-imx219-mipi-ff-nano/ 

maybe a usb camera could work also?

YaBr11, 2020-06-17:
You can use any camera that is compatible with the Jetson Nano for the Jetracer.

closed 2022-07-18 by jaybdub