NVIDIA-AI-IOT/jetracer #60: Basic motion problems JetRacer

Traxan, 2020-05-03:
Hello everyone,

I had a problems with basic_motion notebook.
1. During calibration no matter what numbers I put in car.steering_gain and car.steering_offset my servo does not turn to right. Maximum motion to the right is natural straight run position (which should be on '0' value). Maximum left motion touches chassis and it looks like it would turn more down if there is not a chassis plate.
2. Second problem is about throttle. Contrariwise to the instinctive thinking, when I put negative value the car moves forward and opposite with positive value it moves backward.

Thank you in advance for any hints.
Greetings

krissdap, 2020-05-05:
For question 2, you can set throttle_gain to negative value, that way the throttle should behave in opposite direction.

cscscp, 2020-06-03:
I also encountered the same two problems. My workaround for problem 1 was detaching the servo wheel from the servo and reattaching it after setting the car.steering to 0.

ABilityConsult, 2020-06-08:
Basic Motion.¨
Problem 1: Throttle works only with negative values (not gain, but throttle)
I have not yet managed it to reverse.
Teleoperation.
Problem 1: The steering works, but here the throttle does not respond. Try to negate, but no response

YaBr11, 2020-06-17:
I had the same problem with the "wrong" direction of rotation of the motor.
Besides setting the throttle gain to a negative value, you can also switch two cables from the brushless motor to electrically change the direction of rotation. For me this was the better fix, because my ESC internally reduced the reverse throttle. This led to a much faster reverse speed, than the normal speed with the software fix.

With some ESCs you can also change the direction of rotationin the ESC settings, but I haven't tried that yet.

DeepLJoerg, 2021-03-19:
Please note the throttle is directly mapped to the RC car.  When the car is stopped and a negative throttle is set, it will reverse.  If the car is moving forward and a negative throttle is set, it will brake.

pbalbastre, 2021-04-14:
> I also encountered the same two problems. My workaround for problem 1 was detaching the servo wheel from the servo and reattaching it after setting the car.steering to 0.

How do you do this? I'm having the same problem, throttle works fine but steering only moves a little bit the first time and after that is not moving anymore.

DeepLJoerg, 2021-04-14:
> How do you do this? I'm having the same problem, throttle works fine but steering only moves a little bit the first time and after that is not moving anymore.

I have had the same problem. If your steering does not work anymore, what is the display at the Jetracer showing? 
Mine went black, so the servo must have been crashed. I detached the servo and no problem occurred - e.g. motherboard works. 
A new servo solved the problem, no everything is fine. 

Please consider that the servo may be corrupt.

pbalbastre, 2021-04-14:
> I have had the same problem. If your steering does not work anymore, what is the display at the Jetracer showing?
> Mine went black, so the servo must have been crashed. I detached the servo and no [...]

The steering moves in C++ but not in Python.

DeepLJoerg, 2021-04-15:
> The steering moves in C++ but not in Python.
What is the OLED display at the Jetracer saying in Python and in C++. Any difference?

pbalbastre, 2021-05-04:
Sorry for the late reply. The screen is showing usual info both in Python and C++. No difference.