NVIDIA-AI-IOT/jetracer #63: Can not trainning ?

tjyuanxi, 2020-05-13:
Hi,
     I have take >200 images ,but only twice circles。
    Then I clicked train , 1 epochs or 8 epochs or 10 epochs . I cannot complete progress, it always blocked. Like  the picture below.

[image]
     help me!

tjyuanxi, 2020-05-13:
I set it to 10W mode, it blocked too.
I see meminfo : 3.3G/4G

jasonlililililili, 2021-01-07:
I have encountered the same problem and turns out that there is an image that corrupted (size = 0) and that stucks the training progress from running. You may want to check out the size of each image to see if there are any abnormality.

closed 2022-07-18 by jaybdub