NVIDIA-AI-IOT/jetracer #64: Upgrade to Jetson NX

dsvua, 2020-05-14:
I would like to upgrade to Jetson NX dev board. However, it requires 19v power and I am not sure how to proceed. Is it possible to document upgrade path to Jetson NX and what are our hardware options? Should we use Nano board instead of NX dev board? I would like to use Jetson NX board as it has build in wireless separate M2 slot for SSD and I do not need to share USB port between SSD hard rive and Intel Realsense D435i camera.

dsvua, 2020-05-15:
I guess, upgrade to 3S Lipo battery will work, but will require motor and ESC replacement as well as adding some power split (between ESC and Jetson) board. 3S Lipo battery has voltage 9-12v and, technically, should be enough. Is there a better way?

YaBr11, 2020-06-17:
I also evaluated different power supply options when upgrading to the Jetson Xavier NX.
First thing that came to my mind was also using the Lipo battery, but you already mentioned the drawbacks. Another problem with this approach is the completely missing BMS. So neither the Jetson is protected from overcurrent/overvoltage nor is the battery protected from deep discharge in case you forget to turn off the Jetson.

But there are Powerbanks designed for Laptops that can deliver 12 V and over 60 W. That's the solution I'll go with.

maximveksler, 2020-08-07:
Not an electrical engineer. Interested in building the hardware and playing with the Model.

Would appreciate an upgrade path to Xavier NX

outbackdingo, 2020-09-22:
the xavier can run from 12V also... you dont need 19v.... add a DC-DC NUC power supply / openups2 from mini-box.com

gmanwork, 2020-10-24:
folks, i have been living through the Xavier upgrade for past few months. the sd card image doesn't boot (probably because it is meant for Nano). and manual install also have a bunch of issues due to different jetpack/libraries and almost every module being preinstalled with deprecated tools. i'm still building stuff from source and not gotten to the end of the tunnel yet... 
anywhoooo... the power bank thing is another thread in my story. i hope some of my learning can help others... 
the 5.2mm barrel jack on Xavier is a standard interface and any power bank meant for laptops with same jack or converter would work since most all laptop PSUs are 19v and 3A or more so the power banks for laptops are quite suitable for Xavier. with that said, most of the power banks available are quite expensive ($100 or more). 
one cheaper alternative is to use battery packs that take 4 or more lipo cells (you have to buy the enclosure and cells separately). many available online ([link](https://www.ebay.com/sch/i.html?_from=R40&_trksid=p2334524.m570.l1313&_nkw=19v+power+bank&_sacat=0&LH_TitleDesc=0&_osacat=0&_odkw=battery+pack+12v)). the Cons for this alternative is weight of the package might be 1.2-2lb or more. 
the only suitable but not readily usable option i found was Dell power companion PW7015 ($45 at Staples). it provides 19v, with 12Ah capacity / 65W and around 1lb weight. problem with this one is the stupid dell propitiatory center pin interface which is used for communication from laptop to PSU/power bank to tell it how much volts are needed (and to not charge laptop battery if the PSU / source is below the laptop's expected wattage rating). the power bank shuts down after 30sec. 
i'm working on a hack to spoof it by sending the expected signals (1-wire - ATTINY85 board) but no luck yet...

just wanted to share all my research so we can collectively move forward :)

closed 2022-07-18 by jaybdub