NVIDIA-AI-IOT/jetracer #66: Jetracer Built Using Waveshare AI Kit Not Moving Forward

ArjunGupte1, 2020-06-22:
Hello,
My Waveshare Jetracer Kit (found here: https://www.waveshare.com/wiki/JetRacer_AI_Kit) is not responding to the car.throttle command in the basic_motion jupyter notebook. It is responding fine to the steering commands sent to the servo at the front, but regardless of the throttle value/gain sent to the rear motors, they are not spinning. 

Help on how to debug or suggestions for how to address this problem would be appreciated.

Thanks

openbsod, 2020-06-23:
Hello.

1:1 same problem with Waveshare Jetracer Kit with jetson nano rev A02. Steering works fine, throttle does not work.

Every notebook start 

`>>> from jetracer.nvidia_racecar import NvidiaRacecar
>>> car = NvidiaRacecar()
`

produce these errors

`Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: pio timed out addr: 0x40 tlen:0 rlen:0
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: --- register dump for debugging ----
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: I2C_CNFG - 0x22c00
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: I2C_FIFO_CONTROL - 0xe0
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: I2C_FIFO_STATUS - 0x800080
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: I2C_INT_MASK - 0x7c
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: I2C_INT_STATUS - 0x2
Jun 23 10:29:14 jetson kernel: tegra-i2c 7000c400.i2c: i2c transfer timed out addr: 0x40
`

Thank you in advance.

openbsod, 2020-06-23:
Been there with same issue.

Somehow I found myself with https://github.com/NVIDIA-AI-IOT/jetracer installed instead of  https://github.com/waveshare/jetracer

After proper software was installed troube had gone. Throttle works fine for now.

ArjunGupte1, 2020-06-23:
Thanks for the response. I will try making the change you had suggested and see if the throttle works then. But how did you uninstall the original GitHub repository from the Jetson Nano and then reinstall the new one?

openbsod, 2020-06-23:
I didnt do that.

cd ~/jetracer

```
git remote -v
origin	https://github.com/nvidia-ai-iot/jetracer (fetch)
origin	https://github.com/nvidia-ai-iot/jetracer (push)
```
jetson@jetson:~/jetracer$ **git remote set-url origin https://github.com/waveshare/jetracer**
```
jetson@jetson:~/jetracer$ git remote -v
origin	https://github.com/waveshare/jetracer (fetch)
origin	https://github.com/waveshare/jetracer (push)
```

and sudo python3 setup.py install

chealyte, 2020-06-30:
Hi guys sorry to hijack topic but it seems we got some users of waveshare here. I am really a noob when come to this AI and python.

i am in same problem I could not get car to move forward. I use waveshare repo. I try teleoperation however I can only get it to turn left and right while the car stand still I could not get it to move ..

mipsan, 2020-07-06:
@ArjunGupte1, 

Did you resolve this issue?

My JetRacer also responds for steering controls but does not for throttle.

ArjunGupte1, 2020-07-06:
Yes, sorry for the late response but after using openbsod's fix it worked. You need to use the forked waveshare repository, not the main Nvidia one. After following his steps remember to close and restart the kernel manually in jupyter notebooks before running the new code (or else your old code will keep running and the new code won't get deployed to the jetson nano).

mipsan, 2020-07-07:
Thansk, @ArjunGupte1,

I tested as you said and got same result.
I installed from jetcard to jetracer as here: https://www.waveshare.com/wiki/JetRacer_AI_Kit

When I assign some value to car.throttle, Jetracer makes some noise.
And clear by zero, noise is disappeared. 

Now, I'm gonna try from the first, microSD image file again.

mipsan, 2020-07-07:
@ArjunGupte1 

Nothing! Even I did microSD image job. 
Do you have any info about this DC gear motor? e.g vendor 

It does not move with direct power feeding by power supply. 
12V.

mipsan, 2020-07-07:
@ArjunGupte1 

I found out what's wrong. 
DC Geared motor itself!
Spin motor shaft couple of minutes by hand without power feeding.
And it runs -_-;;

Very embarrassed! 

Thanks,

FortyLuck, 2021-06-05:
> Been there with same issue.
> 
> Somehow I found myself with https://github.com/NVIDIA-AI-IOT/jetracer installed instead of https://github.com/waveshare/jetracer
> 
> After proper software was installed trouble had gone. Throttle works fine for now.

This approach works for me, thanks!

mipsan, 2021-06-07:
https://www.youtube.com/watch?v=5ZMW_KMenio&t=3s

Short video clip here.
I made s short guide book which includes JetRacer, OpenCV and Jetson specific deep learning examples.
With this, to my surprise, some undergraduate guys made this !

TigroTigro, 2021-06-08:
Any ideas what else may be causing problems with Waveshare? I've tried the solution provided by @openbsod but to no avail, sadly :/ Whether it be on the Nvidia or the Waveshare branch, I can run the movement commands with no errors, I can print values of the car's parameters too but it still does not make it move:

```
>>> from jetracer.nvidia_racecar import NvidiaRacecar
>>> 
>>> 
>>> car = NvidiaRacecar()
>>> car.steering = 0.3
>>> print(car.steering_gain)
-0.65
>>> print(car.steering_offset)
0.0
>>> car.throttle = 0.2
>>> print(car.throttle_gain)
0.8
>>> car.throttle_gain = 0.5
>>> 
```

I've checked the connections a couple times and everything seems perfectly as per the instructions (I have the AI Pro kit). The only thing I did _not_ do was connecting the Wifi/Bluetooth module cause in my 2GB Nano there is no extension card slot under the radiator like in the other versions so I bypassed that by having a wifi adapter plugged into the USB port of the Nano - and it works perfectly fine establishing the connection. The only thing that baffles me is that the onboard LCD of the Waveshare board (the one that also holds the batteries) does not ever light up - it'd be weird for it to have power problems considering that the power is passed to the Nano so I don't know whether it plays a role here but I'm mentioning it just in case. Thoughts? :(

closed 2022-07-18 by jaybdub