NVIDIA-AI-IOT/jetracer #68: Out of memory when training with other models

RogerZ1128, 2020-07-03:
Hi，
  I've successfully run the training and roadfollowing with default resnet18 and I noticed that there are some more models to choose. For some need, I'd like to try other models that take less system memory. As far as I know, Alexnet and squeezenet are lighter than resnet18. However, when I run training with these models, system will be out of memory and crashed. What's the problem？Did I miss some settings?
  Thanks.

chealyte, 2020-07-05:
hi RogerZ1128 is there any tip on how to get this road following work? i try mine and i could not get it to work properly ..

RogerZ1128, 2020-07-06:
> hi RogerZ1128 is there any tip on how to get this road following work? i try mine and i could not get it to work properly ..

What's the matter?

chealyte, 2020-07-07:
I mean how you get it to follow road properly with resnet18? I could not get it to do work properly.

tankhunter8192, 2020-07-14:
try to higher the amount of swap memory. Easiest way -> jtop/jetsonstats