NVIDIA-AI-IOT/jetracer #72: Cannot get Teleoperations to work with Gamepad

samaujs, 2020-08-27:
Hi,

  I have managed to obtain GamePad index (=0) but when I execute the following codes in cell 4 of “teleoperation.ipynb” :
—
left_link = traitlets.dlink((controller.axes[0], 'value'), (car, 'steering'), transform=lambda x: -x)
right_link = traitlets.dlink((controller.axes[5], 'value'), (car, 'throttle'), transform=lambda x: -x)
—

resulting in “IndexError: tuple index out of range“ error.  The Gamepad dongle is connected on the remote connecting machine (Mac Catalina or Windows 10) using Google Chrome browser.  There are corresponding updates in axes values with controller interactions (http://html5gamepad.com).

Can someone share some insights how to move forward?  Thanks.

samaujs, 2020-08-30:
This problem has been resolved with "sudo apt install python3-ipywidgets".  Hope it will be useful to those encountering the same problem.

closed 2020-08-30 by samaujs

limch, 2022-07-21:
controller = widgets.Controller(index=1)  --> index=0