NVIDIA-AI-IOT/jetracer #73: the study of collision avoidance about Jetracer

joker-wutong, 2020-09-01:
Has anyone experimented with obstacle avoidance projects on jetracer, just like the collision_avoidance project on jetbot

jaybdub, 2020-09-01:
Hi @joker-wutong ,

Thanks for reaching out!

We actually did play with this a bit, but given the dynamics of JetRacer, the challenge 
is a bit different.  Here's a few i've experimented with

## Method 1 - Hard stop

You could certainly do an image classification approach with JetRacer to accomplish something like ``stop`` / ``go`` but you'd likely want to leave a larger collision buffer, since JetRacer moves faster.   This may be a good safety net for any other approach.

## Method 2 - Soft Stop

Another approach is to actually solve this by just redefining how you collect data using JetRacer's road-following example.

For example, say you want to follow a strict path and slow down based on how close you are to any object in that path.  You could then define JetRacer's target point as the farthest point along the desired path *up until* an object intersects the path.  Because the camera is at a fixed angle and position from the ground plane, you could later determine how far you are away from the object based on the coordinates of that point in the image.

[image]

## Method 3 - Go Around

Similar to method 2, but re-defining in yet another way. If you wanted to "avoid" objects, you could just re-imagine the ideal trajectory as one that moves around the object.  You then set the target point as one as far along that path such that if you were to travel straight to the target point, you would not hit the object.

[image]

## Notes

For any of these methods it would help to have the speed of the vehicle for a good controller.  But a heuristic might get you pretty far.  I've found these approaches actually produce some pretty impressive results without adding much complexity.

Please let me know if this helps or you have any questions.  

Best,
John

joker-wutong, 2020-09-01:
Thank you very much for your ideas, I will do it, if I encounter any problems, I will post them

closed 2022-07-18 by jaybdub