NVIDIA-AI-IOT/jetracer #75: 'Tensor' object has no attribute 'ndim' error in road following

abuelgasimsaadeldin, 2020-09-10:
I am using Jetracer by Waveshare and when implementing the following codes in road_following.ipynb: 

from torch2trt import torch2trt

data = torch.zeros((1, 3, 224, 224)).cuda().half()

model_trt = torch2trt(model, [data], fp16_mode=True)

I receive the following error:

AttributeError                            Traceback (most recent call last)
<ipython-input-3-dd315960311d> in <module>
      3 data = torch.zeros((1, 3, 224, 224)).cuda().half()
      4 
----> 5 model_trt = torch2trt(model, [data], fp16_mode=True)

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/torch2trt.py in torch2trt(module, inputs, input_names, output_names, log_level, max_batch_size, fp16_mode, max_workspace_size, strict_type_constraints, keep_network, int8_mode, int8_calib_dataset, int8_calib_algorithm, int8_calib_batch_size, use_onnx)
    500             ctx.add_inputs(inputs, input_names)
    501 
--> 502             outputs = module(*inputs)
    503 
    504             if not isinstance(outputs, tuple) and not isinstance(outputs, list):

/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py in __call__(self, *input, **kwargs)
    489             result = self._slow_forward(*input, **kwargs)
    490         else:
--> 491             result = self.forward(*input, **kwargs)
    492         for hook in self._forward_hooks.values():
    493             hook_result = hook(self, input, result)

/usr/local/lib/python3.6/dist-packages/torchvision/models/resnet.py in forward(self, x)
    153         x = self.maxpool(x)
    154 
--> 155         x = self.layer1(x)
    156         x = self.layer2(x)
    157         x = self.layer3(x)

/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py in __call__(self, *input, **kwargs)
    489             result = self._slow_forward(*input, **kwargs)
    490         else:
--> 491             result = self.forward(*input, **kwargs)
    492         for hook in self._forward_hooks.values():
    493             hook_result = hook(self, input, result)

/usr/local/lib/python3.6/dist-packages/torch/nn/modules/container.py in forward(self, input)
     95     def forward(self, input):
     96         for module in self._modules.values():
---> 97             input = module(input)
     98         return input
     99 

/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py in __call__(self, *input, **kwargs)
    489             result = self._slow_forward(*input, **kwargs)
    490         else:
--> 491             result = self.forward(*input, **kwargs)
    492         for hook in self._forward_hooks.values():
    493             hook_result = hook(self, input, result)

/usr/local/lib/python3.6/dist-packages/torchvision/models/resnet.py in forward(self, x)
     53             identity = self.downsample(x)
     54 
---> 55         out += identity
     56         out = self.relu(out)
     57 

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/torch2trt.py in wrapper(*args, **kwargs)
    281 
    282             #             print('%s' % (converter.__name__,))
--> 283             converter["converter"](ctx)
    284 
    285             # convert to None so conversion will fail for unsupported layers

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/converters/add.py in convert_add(ctx)
     12     output = ctx.method_return
     13     input_a_trt, input_b_trt = add_missing_trt_tensors(ctx.network, [input_a, input_b])
---> 14     input_a_trt, input_b_trt = broadcast_trt_tensors(ctx.network, [input_a_trt, input_b_trt], output.ndim - 1)
     15     layer = ctx.network.add_elementwise(input_a_trt, input_b_trt, trt.ElementWiseOperation.SUM)
     16     output._trt = layer.get_output(0)

AttributeError: 'Tensor' object has no attribute 'ndim'

Any help would be much appreciated.

TimIWT, 2020-09-10:
I had the same Error and solved it by updating PyTorch to 1.2, but then i can't execute the interavtive_regression anymore.

abuelgasimsaadeldin, 2020-09-10:
Hi @TimIWT 

Thanks for the reply! So is there any other solution that would allow me to be able to execute both road_following as well as interactive regression? I wouldn't want to downgrade PyTorch every time I would like to run the interactive_regression.ipynb. Thanks. Ps. I check my current version of Pytorch and the result was "1.1.0a0+b457266" not sure exactly what this means but i'm guessing its currently PyTorch 1.1?

TimIWT, 2020-09-10:
Yes it is PyTorch 1.1. I am trying to find a solution at the moment and I will let you know as soon as i find one.

abuelgasimsaadeldin, 2020-09-10:
Hi @TimIWT,

Thanks that would be very much appreciated. I will do the same too, as for now I found a temporary solution which is to add: "torch.Tensor.ndim = property(lambda x: len(x.size()))" in the code and that would bypass the error. Still however my road following does not work at all as expected.

TimIWT, 2020-09-10:
What is the problem with the road following notebook? Mine is working but the steering is somehow inverted, so whenever there is a left turn on my track the car goes right.

abuelgasimsaadeldin, 2020-09-10:
Hi @TimIWT 

Yes that is exactly the problem I am facing now, another problem is once I run the the final codes on road following and the tires start to spin, I am unable to make any changes to the steering_gain, steering bias or even car.throttle and so the robot keeps running until I manually have to shut it down by using sudo command. Do you know of any way to bypass this issue? (by adding sliders etc.)
Also how did you manage to solve the direction problem getting it to turn right when it is supposed to and left when it is supposed to instead of the opposite?

Thanks.

TimIWT, 2020-09-10:
About your first problem i am not sure if you can fix it without restarting it (steering_gain and bias seem to be static attributes), but you could use the manual override switch on the RC transmitter to stop the tires as well. 
I will try to find a solution for the direction problem tomorrow, because i need to wait for the battery to charge now. I thought about checking the steering in the basic_motion notebook again. The problem is most probably software-based because it is working when i use the manual control with the RC transmitter.

Also thanks for your solution for the converting problem!

abuelgasimsaadeldin, 2020-09-10:
Hi @TimIWT 

Cool! I will try to look for a solution as well. Thanks!

abuelgasimsaadeldin, 2020-09-10:
Quick update I found the solution to the problem which was to set the steering_gain to negative value for it to be able to react accordingly.

jaybdub, 2020-09-10:
Hi All,

Thanks for pointing this issue out.  

This issue likely surfaced from a change to ``torch2trt`` which references the ``ndim`` attribute.  This change was introduced in this commit

https://github.com/NVIDIA-AI-IOT/torch2trt/commit/48eb79e850734ff4af19f19fa8416cf03aacbfaa

If you try the previous commit of torch2trt, that may allow you to user older versions of PyTorch.  The previous commit is here

https://github.com/NVIDIA-AI-IOT/torch2trt/commit/3ddd70d7cb7074ed204bd77205ab8f6515e06312

While that may be a workaround, it requires reverting to an older torch2trt.  

I'm curious, what was the issue you faced using a later version of PyTorch with the interactive regression notebook? 

Best,
John

TimIWT, 2020-09-11:
Hi All,

@abuelgasimsaadeldin my JetRacer working now! I use the basic_motion notebook to change the throttle whilst running the final code. Thanks again for helping me out!

@jaybdub the problem with the later version of PyTorch was, that the Kernel was restarting as soon as i started the training progress.

Best regards,
Tim

abuelgasimsaadeldin, 2020-09-14:
Hi @TimIWT ,

Nice, I'm glad to hear that. My problem however still persists when trying to run the road_following.ipynb successfully, I have viewed in the interctive_regression.ipynb the ideal path on the live video and everything seems alright but when I try to run it on the road_following.ipynb, the outcome is not the same (It gradually leaves the track whenever it starts to turn) and I have trained multiple images in corners and keep training and still the problem persists.

Any help would be much appreciated :)

jaybdub, 2020-09-15:
As an update, the following PR was merged to torch2trt which should address the original issue here and enable support for older PyTorch versions.

https://github.com/NVIDIA-AI-IOT/torch2trt/pull/415

closed 2022-07-18 by jaybdub