NVIDIA-AI-IOT/jetracer #76: Collision Avoidance and Object Following on Jetracer

abuelgasimsaadeldin, 2020-09-17:
Hi, has anyone tried collision avoidance or object following on the Jetracer? I have worked on those with my Jetbot and would like to do the same with my Jetracer but not sure how. Any help would be much appreciated. Thanks!

abuelgasimsaadeldin, 2020-09-18:
Update:
I have managed to collect collision avoidance data into my Jetracer and have successfully trained the model using codes similar to Jetbot. However when trying to run the live demo on my Jetracer, when running the last line "camera.observe(update, names='value')" I get the following error:

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/dist-packages/jetcam-0.0.0-py3.6.egg/jetcam/camera.py", line 34, in _capture_frames
    self.value = self._read()
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 584, in __set__
    self.set(obj, value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 573, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1205, in _notify_trait
    type='change',
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1210, in notify_change
    return self._notify_observers(change)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1247, in _notify_observers
    c(event)
  File "<ipython-input-12-4d7cee41da0b>", line 18, in update
    car.throttle = -0.4
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 584, in __set__
    self.set(obj, value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 573, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1205, in _notify_trait
    type='change',
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1210, in notify_change
    return self._notify_observers(change)
  File "/usr/local/lib/python3.6/dist-packages/traitlets/traitlets.py", line 1247, in _notify_observers
    c(event)
TypeError: 'int' object is not callable

Anybody knows how to overcome this problem?

chealyte, 2020-09-18:
it would be nice if we can implement it here too but sure if jetson nano have power to process road following and object avoidance at high speed or not since it seem at max processing power already.

closed 2022-07-18 by jaybdub