NVIDIA-AI-IOT/jetracer #77: Road Following turn opposite direction

chealyte, 2020-09-17:
I am using a waveshare kit for jetracer it took me some time to get it to run as I am totally a noob on this. I finally get it to run but it appears to me my car turn left when it suppose to turn right and turn right when it is supposed to turn left.

How do I fix this?

TimIWT, 2020-09-18:
Try to set steering_gain to a negative value, it worked out for me.

chealyte, 2020-09-18:
> Try to set steering_gain to a negative value, it worked out for me.

what is your setting in this area?

this is my setting:

STEERING_GAIN = 0.95
STEERING_BIAS = -0.05

car.throttle = 0.69

TimIWT, 2020-09-21:
My settings are like this:

STEERING_GAIN = -0.7
STEERING_BIAS = 0.00

Changing your steering_gain to -0.95 should solve the problem.

chealyte, 2020-09-21:
@TimIWT Thank massive ... it actually another one that needs to go negative. I keep change bias to negative as instruction said I must read it wrong 

I just use ur setting it to turn correct direction but it does not completely follow the whole loop it at less turn correct and stop make turn .. ( turn half of full ) maybe my training images not good yet it only got 900 pictures .. but I am stepping into the right direction with this, not the crazy car.

TimIWT, 2020-09-22:
@chealyte Unfortunately i am not sure what you mean but, if your car is missing turns, try to increase the steering_gain value (try for example -1.2), if your car is taking the turns too early, try to decrease the steering_gain value (for example -0.6)

chealyte, 2020-09-23:
i got it to work by increase gain and with new training, however, it seems to drive but in that the designed road .. see video ..
i am not sure it is because of bad training or it is more setting to change ..

I trained base on 2200 snapshot of the track with 25 epoch 
STEERING_GAIN = -0.98
STEERING_BIAS = 0.00

https://www.youtube.com/watch?v=79YcqeYRXHw

diamondbarcode, 2021-01-13:
problem solve by just make negative gain

closed 2021-01-13 by diamondbarcode