NVIDIA-AI-IOT/jetracer #80: Question: Pylons along the route

jreinmuth, 2020-10-08:
Hi folks, 

big thank you for this project. As I'm curently trying to train my model and success is limited, just a few questions:
First of all, how do I  manage to get the pylons along the route in my images  - the ones that are shown in your video ?
Are there any tips from your side for training scenarios - e.g. lights, scenery (dos/don'ts), specifics about the surface ?
I would appreciate your help. Thanks in advance.

diamondbarcode, 2020-10-11:
when i trained mine i always try to keep the green circle in between the 2 line of road that we superpose to drive on. This only way i manage to get it successful drive, however it is not accurate it keep drive off road to side ..

jreinmuth, 2020-10-11:
Thanks. I checked the code in the meantime and as I found out, it is only important to click precisely on x-axle, y-axle is not used so far. I managed to run the car two laps (with 170 images for training and another 30 images for evaluating). This did. only work on a plain surface, that has almost same structure and color for the hole track. If I would try this on parquet floor, car will go off-track immediately. Also I think results depend strongly from light - started my outdoor runs at cloudy sky and as sun came out, car suddenly went off track.  I will proceed next week and will add some documentation to my GitHub project

diamondbarcode, 2020-10-12:
this is what i got it to work https://www.youtube.com/watch?v=6Xw9-yzWjnQ .. with 350 images 30 epoch. I am ordering a new camera the one with a light thingy to help see in the dark i help it help indoor training ..

so u say the distance click not help ? ..

jreinmuth, 2020-10-13:
> this is what i got it to work https://www.youtube.com/watch?v=6Xw9-yzWjnQ .. with 350 images 30 epoch. I am ordering a new camera the one with a light thingy to help see in the dark i help it help indoor training ..
> 
> so u say the distance click not help ? ..

Saw that you are using Jetracer from Waveshare, right ? No clue if the y-axle is important for that one. I made mine with the parts list listed here in repo. (First lap completed a few days ago https://www.linkedin.com/posts/juergen-70a4721b0_donkeycar-jetracer-autonomousdriving-activity-6720291417408716800-1iDR)

For x/y-axle, let me describe this a little bit more detailed:

In the original code (in road_following notebook) only values for x  are  used for steering:
...
 output = model_trt(image).detach().cpu().numpy().flatten()
 x = float(output[0])
...
trottle is set to constant value.

diamondbarcode, 2020-10-14:
nice, yes i use wareshare since i am new to this and not much DIYer. I try to get back in this hobbies so i need easy jump start haha .. 

from what i see your now give me thinking something else. So this mean we can make a road with just 1 line ? since it does not need 2 lines or 3 lines to make a road ? i saw you also make split line does it work or car just pick easy road ?

 oh the IR camera modules i buy seem not help .i am testing it and it give weird purplish color of all images .

jreinmuth, 2020-10-14:
Yes, one line is enough, but I guess more lines will help in certain cases, for example when car got to far from center line in a curve and no line is visible in stream of camera anymore. Split line was an ‚accident‘ as my first track design was not feasible , also not in manual mode with remote control. I just delete all the images for that line and restarted training with new line. The car then only takes the new one.

diamondbarcode, 2020-10-14:
I was thinking to get an electrical tap to design a new road but I want to run a lot of color road all over my building haha i try with 1 line and see how it goes. 

Wareshare matt with predesign road i got it to work with a like over 300 images ( really bad at turn )but not accurate at all i could not seem to get it to drive just in the designed road it keeps going off to the outside of the road.

" Waveshare IMX219-160IR Camera for Jetson Nano Infrared Night Vision " is a waste of money it does not help with low light whatsoever unless i implement it wrong. The normal color camera work between then this thing ..

closed 2022-07-18 by jaybdub