NVIDIA-AI-IOT/jetracer #83: JetRacer with Jetson Nano 2gb

seckindroid, 2020-11-02:
Hello,
I want to make this project, and try to run jetracer and  jetbot project at the same time
My question is,
Can I do functions such as road following, obstacle avoid, sign recognition with Jetson nano 2gb and a single camera? Is the device sufficient for these functions, or does the 2gb version run weak for these? Should I get the 4gb version? (as far as I know, the 2gb version doesn't support intel ac 8260)

I'll be glad if you help

jreinmuth, 2020-11-16:
Hi, I have not tried the "light" version of jetson nano, but if I'm using top on jetson while running the playbooks it looks like 2gb is not a good option. So from my POV you should better go for the 4GB version

seckindroid, 2020-11-16:
I understood, thanks for your advice

closed 2022-07-18 by jaybdub