NVIDIA-AI-IOT/jetracer #84: Jetracer doesn't respond to steering and throttle settings in Python code

zhousean, 2020-11-16:
Bought all Tamiya parts and assembled them this summer. Was going through the software set up, but can't really run basic_motion.ipynb. For example, when I run the following lines of code, the car has no response.

NvidiaRacecar().steering = -0.2
car.throttle = 0.5

The car drives fine with the Futaba remote control, forward/reverse and steering. It just doesn't respond to the Python code.

I looked into the hardware setup. RC's CH1/CH2 connect to M1/M2 on the multiplexer and Servo driver CH1/CH2 to S1/S2 on the multiplexer. And RC's CH3 connects to SEL on the multiplexer. If I turn CH3 over 60% on the Futaba RC, sometimes the car will respond to the Python steering/throttle instructions but very inconsistently and in most cases it still doesn't work.

Anyone can shed some light here? I have to admit this is my first time to play with Servo/RC/Multiplexer/Jetson nano. I probably have missed something very basic here.

jreinmuth, 2020-11-16:
Hi,

do you have tried to set other values for steering and throttle in the jupyter notebook as well ?
I also had some issues with my setup (T3PV mit R304SB and brushed motor). You can try setting CH3 to 100% - this is just the kill-switch if AI is going mad :-), so it is on/off -> 0/100%

zhousean, 2020-11-17:
Thanks @jreinmuth. By "Python code", I meant Jupyter notebook code. I am trying basic_motion.ipynb. For me, setting CH3 to 50% seems to be the sweet spot. If I set CH3 to 0%, T3PV RC takes full control. If I set CH3 to 100%, neither T3PV RC nor Jetson Nano controls the steering and throttle.

krissdap, 2020-11-17:
How do you set CH3 to 60%?
Like @jreinmuth stated, CH3 from the radio supposed to either send full 100 or 0, not something in between.

The multiplexer would either forward signal from Futaba receiver or from the servo driver, depends on that signal.
May be it's some kind of trimming on your radio that makes it work that way?

Or is it possible that the multiplexer is faulty? If you can try connecting from the PWM driver directly to the ESC / Servo, see how that work out.

zhousean, 2020-11-20:
T3PV RC allows me to set CH3 incrementally from 0-100% by pressing + or - button.
I also tried to bypass multiplexer by connecting Servo Motor Driver to ESC/Servo. It behaves very similarly. The car responds to steering setting but in most cases there is a couple of seconds delay. The car doesn't respond to most throttle settings but responds to some randomly. Really puzzling.

jreinmuth, 2020-11-20:
I also had some issues first. Steering was ok, but ESC did not work with the standard setting. Check if ESC mode is set to correct values for brushed or brushless motor. If this is set correct, you can check if you'll get your car running with the values I have set https://github.com/jreinmuth/jetracer/blob/master/jetracer/nvidia_racecar.py
ESC behaviour is not necessarily linear - reverse mode needs higher values to get a movement. 
P.S. Just to check - You set the values , saved them for your configuration on T3PV and then you have this config up and running and you have used the switch on the right side of T3PV to turn remote control off and give control to Jetson, right ?

krissdap, 2020-11-21:
@zhousean thanks for clarifying that up.

For throttle, I think you should try changing throttle_gain and throttle_offset value.
TBLE02S ESC has Forward/Brake/Reverse functions, which makes it respond differently on some commands.

Also, just to confirm the direction, ESC yellow wire goes to motor yellow wire, ESC blue wire goes to motor green wire.

zhousean, 2020-11-22:
Thanks @jreinmuth @krissdap I will try to the throttle_gain and throttle_offset and let you know how it goes.
In terms of wiring, I assume mine is correct because if I set CH3 to 0%, I can drive the car fine with RC, forward, reverse, and steering?

zhousean, 2020-12-03:
tried to change the throttle_gain and throttle_offset values but they didn't do anything to change the behavior.

@jreinmuth was reading your question below in your details. I don't think I am doing all those steps with T3PV. Also the switch on the right doesn't really do anything, i.e. it doesn't turn off/on the RC. Any specific documentation that I should read?

P.S. Just to check - You set the values , saved them for your configuration on T3PV and then you have this config up and running and you have used the switch on the right side of T3PV to turn remote control off and give control to Jetson, right ?

jreinmuth, 2020-12-03:
hi, did you also have seen this comment in my file "#after changing values, re-run setup.py" without this, settings are saved but not active as the old instance is used.

for the settings in T3PV, just press the button with arrow on the right side until "SYS" then check if there is value of -100% set for ch3 then move on with the arrow button until you reach "Copy", there I have stored theses settings to 1 (I think it was pressing + and - together until it beeps (i'm not 100% sure right now, so check operations manual of T3PV). then once again arrow button 1 time and you're back in main menu. now you can toggle the switch on right side to turn on/off "jetson control"

With that settings, you should be able to drive the car either fully managed by RC or by Jetson. If not, check wiring and jumpers

MHaz42, 2021-02-28:
Hi, I have the same problem, my steering and throttle doesn't respond to python code but controller works just fine.
Unfortunatly I haven't bought the same controller as the instructions said because it wasn't available in my country.
Maybe it is because of that ? 
Can someone help me ?

MB

closed 2022-07-18 by jaybdub

MounaSamaali, 2023-08-23:
hello, I'am starting controlling the jetracer for the first time. However Jetracer doesn't respond to steering and throttle values coded in python with these lines : 
car.steering = 0.4
car.trottle =0.2
and also the jetracer dose not respond to the gamepad mouvements, I had theses errors on coding. 
 
_### **[IndexError                                Traceback (most recent call last)
<ipython-input-35-28921d6c799e> in <module>
----> 1 left_link = traitlets.dlink((controller.axes[1], 'value'), (car, 'steering'), transform=lambda x: -x)  # link controller 0.5 * controller axes 0 value to car steering
      2 right_link = traitlets.dlink((controller.axes[5], 'value'), (car, 'throttle'), transform=lambda x: -x)  # link 0.3 * axes 1 value to throttle**_

please any help for controlling the movement of the jetracer,

Thank you in advance 
Mouna SAMAALI 
IndexError: tuple index out of range
]([url](

> url

))