NVIDIA-AI-IOT/jetracer #85: Jupyter Lab widget.Controller instead of RC controller

feldiaz, 2020-11-24:
While trying to dlink the widget.controller to the car.steering values as:
steering = traitlets.dlink((controller.axes[1].value, 'value'), (car.steering, 'value), transform = lambda x:-x)
I do have the following error:
"Each object must be HasTraits, not <class 'float'>"
I had created a __init__ def on racecar.py to assure that steering and throttle are traitlets but no luck there.
I am having issues with my RC controller so I am trying to get the Jupyter Lab widget.Controller to communicate with the Adafruit ServoKit library and be able to generate some data for training.
I am aware that Jetbot manage the dlink between controller and DC motor without issues. is there anything different in Jetracer that prevents that?
I appreciate the help.

mvshalamov, 2020-11-29:
you should to understand how the traitlets library works (https://traitlets.readthedocs.io/en/stable/).

in your case you can do this:

```
import traitlets
from traitlets import HasTraits, observe

class Move(HasTraits):
    ax1 = traitlets.Float(default_value=1).tag(config=True)
   
    @observe('ax1')
    def _observe_ax1(self, change):
        car.steering = change['new']

m = Move()

traitlets.dlink((controller.axes[1], 'value'), (m, 'ax1'), transform=lambda x: -x)
```

feldiaz, 2020-11-29:
Hello Maxim,
Thanks for the info.
Unfortunately solution stills not working.
**All below listed code works:**

_import ipywidgets as widgets
import traitlets
from traitlets.traitlets import link, dlink, HasTtraits, observe
controller = widgets.Controller(index=0)
display(controller)_

To test the dlink feature:

_w=widgets.FloatSlider(value=1, min = -1, max=1, step=0.01)
test_dlink = traitlets.dlink((controller.axes[1],'value'), (w, 'value'), transform = lambda x:-x)_

To test the ServoKit library:
(Car steering servo is positioned depending on the controller position, but only one time. I need to re-run the cell to get the steering servo to a new position)

from adafruit_servokit import ServoKit
steering_servo = ServoKit(channels=16)
steering_arm = steering_servo.continuous_servo[1]
steering_arm.throttle = controller.axes[1].value_

All above code works.

**Below code does not produce any physical movement:**

from jetracer.nvidia_racecar import NvidiaRacecar
car = NvidiaRacecar()
class Move(HasTraits):
    ax1 = traitlets.Float(default_value=1).tag(config=True)
   
    @observe('ax1')
    def _observe_ax1(self, change):
        car.steering = change['new']

m = Move()

direction = traitlets.dlink((controller.axes[1], 'value'), (m, 'ax1'), transform=lambda x: -x)_

Something seems to be broken between widgets.Controller and its ability to connect to the ServoKit library. What is intriguing is that for DC motors, Nvidia Jetbot seems to got it working.

closed 2022-07-18 by jaybdub