NVIDIA-AI-IOT/jetracer #87: AttributeError: module 'torch' has no attribute 'reciprocal_'

Jundor-y, 2020-12-12:
When running `from torch2trt import torch2trt` the following error occures:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-1-3f7e7e224fb7> in <module>
----> 1 from torch2trt import torch2trt

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_unlocked(spec)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_backward_compatible(spec)

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/__init__.py in <module>
      1 from .torch2trt import *
----> 2 from .converters import *
      3 import tensorrt as trt
      4 
      5 

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_unlocked(spec)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_backward_compatible(spec)

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/converters/__init__.py in <module>
     15 from .Linear import *
     16 from .LogSoftmax import *
---> 17 from .activation import *
     18 from .adaptive_avg_pool2d import *
     19 from .adaptive_max_pool2d import *

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_unlocked(spec)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_backward_compatible(spec)

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/converters/activation.py in <module>
      1 from torch2trt.torch2trt import *
      2 from torch2trt.module_test import add_module_test
----> 3 from .unary import UnaryModule
      4 
      5 

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_unlocked(spec)

/usr/lib/python3.6/importlib/_bootstrap.py in _load_backward_compatible(spec)

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/converters/unary.py in <module>
     71 
     72 @tensorrt_converter('torch.reciprocal')
---> 73 @tensorrt_converter('torch.reciprocal_')
     74 @tensorrt_converter('torch.Tensor.reciprocal')
     75 @tensorrt_converter('torch.Tensor.reciprocal_')

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/torch2trt.py in tensorrt_converter(method, is_real, enabled, imports)
    587         module, module_name, qual_name = importlib.import_module(method.__module__), method.__module__, method.__qualname__
    588 
--> 589     method_impl = eval('copy.deepcopy(module.%s)' % qual_name)
    590 
    591     def register_converter(converter):

/usr/local/lib/python3.6/dist-packages/torch2trt-0.1.0-py3.6.egg/torch2trt/torch2trt.py in <module>

AttributeError: module 'torch' has no attribute 'reciprocal_'
```

It's running the `jetcard_v0p0p0.img ` image from the NVIDIA JetRacer GitHub repo and the setup instructions to install additional packages of jetcam, torch2trt and jetracer were followed. 

Hardware is a WaveShare JetRacer with a Jetson Nano B01

Is this is a bug and is there a workaround for this AttributeError?

Jundor-y, 2020-12-13:
Solved by installing torch1.2 as suggested here: https://github.com/NVIDIA-AI-IOT/jetracer/issues/27

closed 2020-12-13 by Jundor-y

ilhmfahmi, 2021-01-16:
heyy, how to installing torch 1.2?