NVIDIA-AI-IOT/jetracer #88: ImportError: libcudart.so.10.2: cannot open shared object file: No such file or directory

Jundor-y, 2020-12-13:
```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-85080642f8f2> in <module>
----> 1 import torch
      2 import torchvision
      3 
      4 CATEGORIES = ['apex']
      5 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in <module>
     79 del _dl_flags
     80 
---> 81 from torch._C import *
     82 
     83 __all__ += [name for name in dir(_C)

ImportError: libcudart.so.10.2: cannot open shared object file: No such file or directory
```

**Software**:
jetcard_v0p0p0.img from https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md
torch-1.2 (manually updated from torch-1.1 following this advice: https://github.com/NVIDIA-AI-IOT/jetracer/issues/27)
cuda-10.0 (was preinstalled)
**Hardware**:
Jetson Nano B01

Jundor-y, 2021-01-01:
Used the following method to get the Waveshare JetRacer up and running:

1) Downloaded chinese image posted here: (https://forums.developer.nvidia.com/t/jetracer-python-examples/155119/21)

2) Reinstalled Jetcard and JetRacer packages, as instructed under step 5 from here: (https://www.waveshare.com/wiki/JetRacer_AI_Kit)

ilhmfahmi, 2021-01-26:
How to update torch-1.2.0?