NVIDIA-AI-IOT/jetracer #90: torch2trt no attribute for reciprocal_

ilhmfahmi, 2021-01-26:
i have some error when trying road folowing example on jetracer 
in https://github.com/NVIDIA-AI-IOT/jetracer/issues/27 said that version 1.2 would resolve this issue
so how to upgrade torch-1.2.0? 
please help, thanks.
[image]

vincentandreas, 2021-01-30:
Hi @ilhmfahmi . I think you should try to install torch library from this link https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-7-0-now-available/72048

abritten, 2021-02-01:
@ilhmfahmi I tried this and it worked earlier this month:

wget https://nvidia.box.com/shared/static/06vlvedmqpqstu1dym49fo7aapgfyyu9.whl -O torch-1.2.0a0+8554416-cp36-cp36m-linux_aarch64.whl

pip3 install numpy torch-1.2.0a0+8554416-cp36-cp36m-linux_aarch64.whl --user

But now its not working. Try it anyways.

closed 2022-07-18 by jaybdub