NVIDIA-AI-IOT/jetracer #91: The problem of not booting when burning jetcard image to sd card

Choikyungho9, 2021-02-04:
jetcard (OS) state seems odd.

I put jetcard_v0p0p0.zip onto the sd card and attached it to the jetson nano.

when i burn jetcard to SD card, there is a problem that booting is not possible.

Only ndivia is displayed and it does not boot.

Are there other versions of jetcard?

please help me.

jsopdebeeck, 2021-02-07:
Same here. The image is not working.

Rafipaulino, 2021-02-17:
Has somebody fixed this problem? I have the same.

TomasVraspir, 2021-02-18:
I have the same problem. Does it have a solution?

jsopdebeeck, 2021-02-18:
Hello, I solved the issue by replacing the CPU unit with an other one (same , B01), then everything started well. Or check that your CPU unit is 100% clean and well plugged.

grahil247-code, 2021-02-22:
Not sure that would fix the issue. Swapping out another CPU from the same board shoe that this is possibly a firmware / driver issue. I have the same problem so something must have changed for this not to work. Hope it is resolved soon either way!

hkstemclub, 2021-03-04:
Same problem here, half year before I use the same image in old jetson nano B01 board and it was working well....now I use it in the new board it doesn't work.....

legonger, 2021-03-06:
I have the same problem.
When I put another img file (e.g.  "jetson-nano-jp451-sd-card-image.zip") on to the same SD card, it boot normally.
But I put "jetcard_v0p0p0.zip" on to the same SD card, it does not boot.
Only boot logo "NVIDIA" is displayed repeatedly...

zinwalin, 2021-03-08:
> I have the same problem.
> When I put another img file (e.g.  "jetson-nano-jp451-sd-card-image.zip") on to the same SD card, it boot normally.
> But I put "jetcard_v0p0p0.zip" on to the same SD card, it does not boot.
> Only boot logo "NVIDIA" is displayed repeatedly...

same here.  I have two B01 board, the same image doesn't work the same behavior even specs are same.

arickchan, 2021-03-10:
I have tried to buy a new A02 board last week and it's work. So I think the problem is the capability between Jetcard image and the new lot if the B01 boards.

tokk-nv, 2021-03-10:
Hi all,
Can anybody take the debug console log (by following a method introduced in [this Jetsonhacks article](https://www.jetsonhacks.com/2019/10/10/jetson-nano-uart/)) and share with us?

zinwalin, 2021-03-15:
> Waveshare did send me another image that booted but has an old jetpack 4.2
> on it. It booted, though, so it is with the image and not our devices. I
> have flashed JP4.4 which boots fine, but some install error and difficult
> to install, for example, Jupyter labs and connect remotely and an older
> version of TensorFlow 1.15.x. Finally, have a working image for the jet
> racer that seems to work, and I can control the car, but... The Jetracer
> Pro boots but errors cannot see the camera, expansion board and reports no
> device at i2c 0x60. I can see something on i2c 1, but still, it will not
> work.
>
> I have 1x A02 and 2xB01 boards: the Jetracer and the Jetracer Pro.

They updated the image last weekend.

riaz, 2021-08-13:
Looks like its not resolved yet!

diamondbarcode, 2022-05-04:
is anyone with a sd image that work can be share ? ..

arickchan, 2022-11-25:
@tokk-nv 

Hello everyone,

I tried the new image with new version of jetpack, It works perfect in the training notebook.

But when I try the road following program, error message comes with the first code I run:

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-2-85080642f8f2> in <module>
----> 1 import torch
      2 import torchvision
      3 
      4 CATEGORIES = ['apex']
      5 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in <module>
    186     # See Note [Global dependencies]
    187     if USE_GLOBAL_DEPS:
--> 188         _load_global_deps()
    189     from torch._C import *
    190 

/usr/local/lib/python3.6/dist-packages/torch/__init__.py in _load_global_deps()
    139     lib_path = os.path.join(os.path.dirname(here), 'lib', lib_name)
    140 
--> 141     ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
    142 
    143 

/usr/lib/python3.6/ctypes/__init__.py in __init__(self, name, mode, handle, use_errno, use_last_error)
    346 
    347         if handle is None:
--> 348             self._handle = _dlopen(self._name, mode)
    349         else:
    350             self._handle = handle

OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

Any idea to solve the problem? thanks everyone help me.

Rick