NVIDIA-AI-IOT/jetracer #94: jetracer basic motion error

jackey42, 2021-03-02:
[image]
[image]
[image]

sudo pip3 install --upgrade pip
sudo pip3 uninstall Adafruit-PureIO
sudo pip3 install Adafruit-PureIO

We already tried this, but the error has not solved.

Please help ;n;

jackey42, 2021-03-10:
> Hi jackey42,
> Can you tell me what vehicle hardware you are using?
> (Latrax? Tamiya? or Waveshare JetRacer? Waveshare JetRacer Pro?)

Hi~!
I'm using Waveshare JetRacer.

zinwalin, 2021-03-15:
> Hi~!
> I'm using Waveshare JetRacer.
>  
> -----Original Message-----
> From: "Chitoku YATO"<notifications@github.com>
> To: "NVIDIA-AI-IOT/jetracer"<jetracer@noreply.github.com>;
> Cc: "jackey42"<jenny_jung@naver.com>; "Author"<author@noreply.github.com>;
> Sent: 2021-03-10 (수) 16:19:42 (GMT+09:00)
>  
> Hi jackey42,
> Can you tell me what vehicle hardware you are using?
> (Latrax? Tamiya? or Waveshare JetRacer? Waveshare JetRacer Pro?)

use waveshare image，and deploy patches for specific hardware.

tokk-nv, 2021-03-19:
Hi jackey42,

As zinwalin suggested, please check out the Waveshare image which they also posted in this issue here.
https://github.com/NVIDIA-AI-IOT/jetracer/issues/96

openbsod, 2021-04-26:
Hello jackey42

Met same errors with JetRacer Pro. So my conditions are:

- JetRacer Pro https://www.waveshare.com/wiki/JetRacer_Pro_AI_Kit

- jetcard_nano-4gb-jp451 https://doc-0c-bc-docs.googleusercontent.com/docs/securesc/oca2mgdh5rj4h8hbtk0do6b37cs03g8n/rqmq8oajntfduih3euka329cmkam653k/1619391975000/08570363794112935001/15224068068018828316/1MX-z7ZCPvUzpN3nGhfZMAgENtK6VnBdh?e=download&authuser=0&nonce=4tsnrhoii3rl4&user=15224068068018828316&hash=8me5382sk3s71098iksrdem94gv256n9

- Jetson Nano (A02) 4GB

- branch ws/pro https://github.com/waveshare/jetracer/tree/ws/pro

With this branch I can get basic motions notebook about fully working execpt backward motion. It can move forward like normal jetracer do, but moves backward with 'donkeycar' scheme 'Reverse on RC cars is a little tricky because the ESC must receive a reverse pulse, zero pulse, reverse pulse to start to go backwards.' as described at https://docs.donkeycar.com/guide/calibrate/

BR, Ed.

closed 2022-07-18 by jaybdub