NVIDIA-AI-IOT/jetracer #96: Pre-built image (Jetpack4.5) for Waveshare JetRacer Pro AI Kit and JetRacer AI Kit

waveshare, 2021-03-13:
We make a Jetracer Pro image based on Jetpack4.5. If you use Jetson Nano Developer Kit (B01) and the Waveshare JetRacer Pro AI, Kit. You can download this image and test it.
link: https://drive.google.com/file/d/1bgCAUJ9m16g5FGuYgKy3WmGI-pGjPXN0/view?usp=sharing

If you have the JetRacer AI Kit. Please switch the Jetracer repository and reinstall with the following commands:
```
cd jetracer
git checkout master
sudo python3 setup.py install
sudo reboot
```

korzen, 2021-03-26:
@waveshare  sorry, but it is still unclear to me what I should do to get the JetRacer AI Kit (non-pro with Nano A01) up and running. I have burned a fresh JetPack 4.5.1 and manually installed the JetCard from 4.5.1 branch. The thing is that the Jupyter is not starting but that is different story. Which repo/branch should I use to get the RaceCar working with 4.5 update. Thanks

DeepLJoerg, 2021-04-19:
> @waveshare sorry, but it is still unclear to me what I should do to get the JetRacer AI Kit (non-pro with Nano A01) up and running. I have burned a fresh JetPack 4.5.1 and manually installed the [...]

Did you download the right image? There is one for 2GB and one for 4GB.

diamondbarcode, 2022-05-04:
i am 2nd on this, where is pre-made sd card image for none pro versioin ? ..

waveshare, 2022-05-06:
@diamondbarcode 
The non pro version uses the same image as the pro version. The only different is that you need to go into the jetracer directory and swich the batch to master and re-install it.
Please check our wiki for details.

hemnsue, 2023-11-21:
hello waveshare i need your help, kindly reply please ,in my waveshare jet racer ai kit (non pro version) the rear motors are not running they don't respond to the basic motion when i set car.throttle=1 or -1 or anyother value between -1 to 1. My servo motors are running well and good i just needed help in this problem . The car.throttle_gain has been set to 0.8 , but i don't get any kind of response from the dc motors when i run it, servo motors are running good, i have the Jetson Nano B0-1 board and all my connections are correct i have rechecked them many times and also when i'm normally power the motors to check if they are fine using a battery they are running good so there is no problem with the motors in general . Please help . It will be really helpful as i really want to explore and learn about autonomous driving and do some personal projects.