NVIDIA-AI-IOT/jetracer #99: A few small setup issues with new Jetpack 4.5 release

zlite, 2021-03-27:
Looks like you need "sudo" in from of those git clone commands (sudo does indeed resolve these issues). 

[image]
[image]

Also, if you happen to have a monitor attached while doing the bootup, you'll note this error:
[image]

YaBr11, 2021-03-28:
Same for me.
Another issue I detected is that PyTorch is not working inside the JupyterLab environment.
When I try to import torch or torchvision I get: `OSError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block`
In the terminal using the python3 environment, it works fine on the other hand.

TCIII, 2021-05-28:
@zlite,

I originally [reported](https://forums.developer.nvidia.com/t/issues-setting-up-jetracer-with-the-latest-release-image/178952) the issues you have experienced and more on the NVIDIA Developers Nano Forum on May 25th.

They acknowledged the fact that there are issues with the latest release, however I don't think that they are in any rush to fix them as the it has been two years since the Jetracer software image was updated until now.

TCIII, 2021-05-31:
Observation with the latest Jetracer release for the Nano 4GB B01:

If you successfully get the latest Jetracer release installed you will find that you cannot save any of the Jetracer Notebooks as permission is denied.

TCIII, 2021-05-31:
Here is a workaround for the installation of the latest Jetracer release that has worked for me:

1) Burn [JP 4.5.1](https://developer.nvidia.com/embedded/jetpack) onto a micro SD card and complete the setup using the Ubuntu Desktop.
2) From a SSH terminal run sudo apt-get update -y followed by sudo apt-get upgrade -y and then a reboot.
3) From a SSH terminal clone and install [Jetcard](https://github.com/NVIDIA-AI-IOT/jetcard) using Option2 Step 2.
4) From a SSH terminal clone and install Jetcam and Jetracer (don't use sudo with clone) per Step 5 in the [Jetracer](https://github.com/NVIDIA-AI-IOT/jetracer/blob/master/docs/software_setup.md) installation guide, but don't clone and install torch2trt as the Jetcard installation has already installed it. 
5) You may get a message that Jupyter needs to build/install additional software when you first run the "basic_motion" notebook, however the install will fail as permission is not granted. So note the build command and use a SSH terminal prefacing the command with "sudo" to get the build to complete.

Following the above steps I found that I could now save the notebooks without a denial of permission.

Comments?

TCIII, 2021-05-31:
> Same for me.
> Another issue I detected is that PyTorch is not working inside the JupyterLab environment.
> When I try to import torch or torchvision I get: `OSError: [...]

When I installed the latest release of Jetracer from scratch (see above) torch and torchvision imported without issue when I ran the Model section of the interactive_regression.ipynb notebook.

TCIII, 2021-06-09:
@zlite,

FYI

NVIDIA has addressed the issues I pointed out and packaged a new image, partly taking my suggested solutions.

zlite, 2021-06-09:
Excellent. Thx!