NVIDIA-AI-IOT/jetracer #101: No response to Basic_motion

sdovhl, 2021-05-30:
Hello everyone,

During running through the basic_motion notebook we couldn’t get any response from the car.
The transmitter and reciever we are using is the Carson Reflex PRO 3 (without LCD) the transmitter als isnt equiped with a manual override switch, when self operating the car, everything works just fine however whenever we try to get to run the car through the basic_motion notebook the car doesnt response on any setting we try  (neither steering nor throttle).

We tried to set the throttle_gain to a negative value, which also didnt give any response. We have also tried to bypass the multiplexer and connect the PCA PWM driver tot he ESC directly which also didnt give any response. 

The transmitter is equiped with reverse switches which reverse the throttle and steering, while driving the car manually setting the reverse throttle switch to reverse mode will get the car to automatically drive in reverse motion without moving the throttle trigger forward backward. Whilst trying to run the Basic_motion notebook during this mode, we couldnt get any response either. 

Thank You in advance for any hints.

sdovhl, 2021-05-30:
We are using the Tamiya jetracer

TCIII, 2021-05-31:
You didn't state what version of Jetracer you are using. 
Are you using the Jetracer latest release on a Nano 4GB B01 or an earlier version?

sdovhl, 2021-05-31:
> You didn't state what version of Jetracer you are using.
> Are you using the Jetracer latest release on a Nano 4GB B01 or an earlier version?

We are using the latest version on a Nano 4GB B01 indeed.

TCIII, 2021-05-31:
> > You didn't state what version of Jetracer you are using.
> > Are you using the Jetracer latest release on a Nano 4GB B01 or an earlier version?
> 
> We are using the latest version on a Nano 4GB B01 indeed.

I assume that you used the latest release image and did not build Jetracer from scratch?
Did you take a look at open issue #99 before doing the Jetracer image install? Issue #99 may explain the issues that you are having with "basic_motion".

sdovhl, 2021-05-31:
> I assume that you used the latest release image and did not build Jetracer from scratch?
> Did you take a look at open issue #99 before doing the Jetracer image install? Issue #99 may explain the [...]

I installed the latest release according to your guide in open issue #99 before installing the Jetracer image, however at step 4 permission is still denied while trying to install Jetcam and Jetracer. Basic motion still gives no response when running it

TCIII, 2021-05-31:
@sdovhl 

When you say that you "installed the latest release" are you talking about the Jetracer latest release image?

If so, that is not what I did to get my installation of Jetracer to work. I did a clean install and setup of of Jetpack 4.5.1 and then cloned and installed Jetcard followed by Jetcam and Jetracer.

sdovhl, 2021-05-31:
I followed your steps in issue #99 from start to finish, are there any more steps i had to do beforehand?
As I downloaded the zip file of jp 4.5.1 and transferred it to a micro SD, after that i followed on from step 2 however at step 4 it still kept on saying permission denied when trying to clone the packages.

TCIII, 2021-06-01:
@sdovhl,

I am sorry to hear that my guide has not worked for you.

Somehow you are doing something differently than I did during the installation process and unfortunately I don't know what it is.

sdovhl, 2021-06-01:
When running sudo apt-get upgrade -y, It said some packages couldnt get the upgrade and said i should try  sudo apt-get upgrade --fix-missing, which i did. During the whole process it asked me to update the software (to ubuntu 18.04). It also asked my a question twice which i had to answer (y/N), once for overwriting jupyiter notebook with default Data or something like that which i rejected (N) and the other question was to install an extra package. 

Just some things that were noticable, do you know if any of these might have caused the problem or is it better for me to just go with an older version of jp if that would even work?

TCIII, 2021-06-01:
> When running sudo apt-get upgrade -y, It said some packages couldnt get the upgrade and said i should try sudo apt-get upgrade --fix-missing, which i did. During the whole process it asked me to [...]

Just run sudo apt-get update -y and then sudo apt-get upgrade -y and then do a reboot. Don't worry about the packages that couldn't be upgraded as they usually can't be upgraded with the NVIDIA version of Ubuntu 18.04.  Just follow my installation guide to the letter and you should have a successful install.

sdovhl, 2021-06-09:
When following your guide, I still couldnt get any response out of the car. After the reboot i still get the same error message shown when viewing it on a monitor. Still cant get any motion going when running basic_motion. 
I have also tried with jetpack 4.5, but still couldnt get any motion going on.

TCIII, 2021-06-09:
@sdovhl,

FYI

NVIDIA has addressed the issues I pointed out and packaged a [new image](https://forums.developer.nvidia.com/t/issues-setting-up-jetracer-with-the-latest-release-image/178952/7), partly taking my suggested solutions.

sdovhl, 2021-06-10:
> @sdovhl,
> 
> FYI
> 
> NVIDIA has addressed the issues I pointed out and packaged a [new image](https://forums.developer.nvidia.com/t/issues-setting-up-jetracer-with-the-latest-release-image/178952/7), partly taking my suggested solutions.

Yes I saw! But isnt the new image just the 4.5.1? I've tried basic motion with their newest image however still cant get any motion when running through Basic_motion

TCIII, 2021-06-10:
@sdovhl,

The Jetracer "image" is not just JP 4.5.1, but includes Jetcard which builds a lot of packages and dependencies if you run it separately from JP 4.5.1.
Is your jet racer a kit or is it a DIY project?

Have you verified that the Nano is communicating with the PCA9685 servo driver module over the I2C bus?
Run the following to command vis SSH to verify that the Nano is communicating with the PCA9685:

sudo i2cdetect -r -y 1

The PCA9685 should appear at hex address 0x40.

If it appears at the correct hex address, then connect the ESC and steering servo cables directly to the PCA9685 and see if you can get basic_motion to operate the car throttle and steering. If that works, then there is something wrong with the multiplexer or the way that you have attached the throttle/steering cables to it. 
If it does not appear at hex address 0x40, then that is where your problem is.

sdovhl, 2021-06-16:
Hi, thank you for your help! We have now solved the issue.
With the new image we tried to bypass the multiplexer (connect the PCA motor driver directly to the ESC) and this time it was succesfull, we were able to run "basic_motion" now. 
We have installed a manual override switch on channel 3 of the transmitter and now it works.

closed 2021-06-16 by sdovhl