NVIDIA-AI-IOT/jetracer #103: JetRacer AI Kit

mipsan, 2021-06-02:
I provided these undergraduate guys with this kit and guide book. 
They finally made JetRacer run on their own track.

https://youtu.be/5ZMW_KMenio 

My guide book covers from the basic examples of JetRacer to OpenCV, deep learning and Jetson specific inference engines as well.

zlite, 2021-06-02:
Nice! Is your guide book public?

mipsan, 2021-06-04:
Yes it is, but written in Korean. 
So Sorry.

ksj20, 2021-07-11:
안녕하세요! 혹시 공개된 가이드북은 어디서 찾을 수 있나요? 감사합니다.

mipsan, 2021-07-12:
@ksj20 
It's not a public material, but is included my Kit. 

Would you wanna get some more info about it?

Thanks,

ksj20, 2021-07-14:
@mipsan Yes, that would be great! Thanks

GilHo-Roh, 2021-08-08:
@mipsan 
I'm also testing similar experiment.
The result of your video is very impressive.
Could I refer to your guidebook by any chance?
Thanks

저도 비슷한 실험을 진행 중에 코멘트 남깁니다.
유튜브의 결과 영상이 정말 인상적이며, 저도 비슷한 결과를 내고 싶은데
어려움을 조금 겪고 있습니다.
혹시 저도 가이드북을 참고할 수 있는 방법이 있을까요?
감사합니다.

ksj20, 2021-08-09:
@GilHo-Roh 
It would be great if you share any updates from @mipsan! Since I didn't receive any more info after my request. Thanks

@mipsan으로부터 연락을 받으신다면 공유해주시면 좋을 것 같아요! 저도 지난번 요청 이후로 따로 연락을 못 받았어서요, 감사합니다

mipsan, 2021-08-10:
@GilHo-Roh @ksj20 
Thanks for your interesting!

I hope to open and share the source tree soon.
The guidebook was made for the customer who buys JetRacer from me.
Thanks,