NVIDIA-AI-IOT/jetracer #104: camera initialization issue everytime switch from train to run road following

annduy, 2021-06-08:
Hi 
I do complete the training then open the road following notebook to run but always always get the following issue 
`---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
/usr/local/lib/python3.6/dist-packages/jetcam-0.0.0-py3.6.egg/jetcam/csi_camera.py in __init__(self, *args, **kwargs)
     23             if not re:
---> 24                 raise RuntimeError('Could not read image from camera.')
     25         except:

RuntimeError: Could not read image from camera.

During handling of the above exception, another exception occurred:

RuntimeError                              Traceback (most recent call last)
<ipython-input-3-05c12066f81a> in <module>
      6 from jetcam.utils import bgr8_to_jpeg
      7 
----> 8 camera = CSICamera(width=224, height=224, capture_fps=65)
      9 #camera.running = True
     10 image = ipywidgets.Image(format='jpeg')

/usr/local/lib/python3.6/dist-packages/jetcam-0.0.0-py3.6.egg/jetcam/csi_camera.py in __init__(self, *args, **kwargs)
     25         except:
     26             raise RuntimeError(
---> 27                 'Could not initialize camera.  Please see error trace.')
     28 
     29         atexit.register(self.cap.release)

RuntimeError: Could not initialize camera.  Please see error trace.`

I have to restart the jetson nano everytime switching from training to running.
Please help.

Thanks,

AN

TCIII, 2021-06-09:
@annduy,

I have seen this error before and it is because the Jetracer training notebook does not stop the camera (camera.stop()) at the end of training like the Jetbot notebooks do. This code fragment stops the camera once it has been started:

mipsan, 2021-06-11:
@annduy 
This error occurs so many times!
You can treat this  error as @TCIII  recommends or run this command :
`$ sudo systemctl restart nvargus-daemon`

annduy, 2021-06-11:
@TCIII & @mipsan: thank you both for great advices. No need to restart jetson nano the time by time now. Thanks and have a nice weekend.
AN

MaxKru, 2022-03-09:
> @annduy,
> 
> I have seen this error before and it is because the Jetracer training notebook does not stop the camera (camera.stop()) at the end of training like the Jetbot notebooks do. This Python code fragment stops the camera once it has been started:
> 
> > def stop(self):
> > if not self._running:
> > return
> > self._running = False
> > self._thread.join()

where exactly should i put this code, i guess in the jetcam/csi_camera.py but dont want to mess up anything,
please be so kind and tell me exactly where to put it.

in jetracer/waveshare fork notebook, the csi_camera gets called thats why i think it belongs in there.

mipsan, 2022-03-10:
@MaxKru,

Try that you think,
IF you're worried, backup the file `csi_camera.py` and modify.