NVIDIA-AI-IOT/jetracer #105: No response to basic motions but no errors either

TigroTigro, 2021-06-08:
I've just installed jetracer on my 2GB Jetson Nano creating the card manually through the Jetcard and Jetracer tutorials and everything seems fine, I assembled my Waveshare Jetracer Pro AI Kit, I mounted Jetson on top of it, the fan is working but when I try to go through any of the basic motions, be it through the provided notebooks or on my own through the terminal, nothing happens. Here's the log:

```
>>> from jetracer.nvidia_racecar import NvidiaRacecar
>>> 
>>> 
>>> car = NvidiaRacecar()
>>> car.steering = 0.3
>>> print(car.steering_gain)
-0.65
>>> print(car.steering_offset)
0.0
>>> car.throttle = 0.2
>>> print(car.throttle_gain)
0.8
>>> car.throttle_gain = 0.5
>>> 
```

As you can see, there are no errors but the car stands still. I don't know how I can force it out of the manual (if that even is the problem) cause it does not even seem to be in it as I have not installed the [wireless module antennas](https://www.amazon.com/Waveshare-AC8265-Wireless-Supports-Bluetooth/dp/B07SGDRG34) (as my Nano is not supporting external extension cards) and am instead using a USB network adapter which provides WiFi to the Nano.

How can I make it work?

s95huang, 2021-06-09:
Hi, I had the same problem before. Take it easy~ Here is the solution:

You need to turn on the RC ESC switch hidden under the left WiFi antenna or near the rear left tire to get it to move.\
It took me a while to find the switch so please look carefully.
If done properly, you should hear a  beep and probably see a red light, meaning that the ESC is on

When turning on the fan, the jetson nano could consume more than 5W which is more than what the battery is designed to supply. \
I suggest you to turn it off.

You should be able to install the Wireless NIC Module by removing the jetson nano compute module (aka the card). \
The USB WiFi dongle should work fine.

Please be careful when playing with the throttle value since mine run off the stairs (It didn't break) ...

TigroTigro, 2021-06-09:
Yes, that was it! Thanks so much @s95huang :D 

> You should be able to install the Wireless NIC Module by removing the jetson nano compute module (aka the card). 
> The USB WiFi dongle should work fine.

Sadly, my version does not feature the connector underneath the module. So while [this](https://docs.nvidia.com/isaac/archive/2019.1/_images/nano-wifi-4.jpg) is what it looks like on many Jetsons, mine has neither that slot nor that screw spot, thus preventing me from installing the module. On a related note, can you use the controller in such a situation or is installing this module crucial for this? Cause right now, I have no way to switch to manual steering and I wonder if that controller needs this module or if it can control the car just fine without it?

closed 2021-06-09 by TigroTigro